orangepi-xunlong/linux-orangepi #5: SSH卡死问题

Longhao-Chen, 2021-05-15:
今早ssh无法登录，在输入密码后就没有了响应，使用 `ssh -v` 得到的信息如下：
```
root@192.168.3.50's password: 
debug3: send packet: type 50
debug2: we sent a password packet, wait for reply
debug3: receive packet: type 52
debug1: Authentication succeeded (password).
Authenticated to 192.168.3.50 ([192.168.3.50]:22).
debug1: channel 0: new [client-session]
debug3: ssh_session2_open: channel_new: 0
debug2: channel 0: send open
debug3: send packet: type 90
debug1: Requesting no-more-sessions@openssh.com
debug3: send packet: type 80
debug1: Entering interactive session.
debug1: pledge: network
debug3: receive packet: type 80
debug1: client_input_global_request: rtype hostkeys-00@openssh.com want_reply 0
debug3: receive packet: type 91
debug2: channel_input_open_confirmation: channel 0: callback start
debug2: fd 3 setting TCP_NODELAY
debug3: ssh_packet_set_tos: set IP_TOS 0x10
debug2: client_session2_setup: id 0
debug2: channel 0: request pty-req confirm 1
debug3: send packet: type 98
debug1: Sending environment.
debug1: Sending env LANG = zh_CN.UTF-8
debug2: channel 0: request env confirm 0
debug3: send packet: type 98
debug2: channel 0: request shell confirm 1
debug3: send packet: type 98
debug2: channel_input_open_confirmation: channel 0: callback done
debug2: channel 0: open confirm rwindow 0 rmax 32768
debug3: receive packet: type 99
debug2: channel_input_status_confirm: type 99 id 0
debug2: PTY allocation request accepted on channel 0
debug2: channel 0: rcvd adjust 2097152
debug3: receive packet: type 99
debug2: channel_input_status_confirm: type 99 id 0
debug2: shell request accepted on channel 0
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: receive packet: type 98
debug1: client_input_channel_req: channel 0 rtype keepalive@openssh.com reply 1
debug3: send packet: type 100
debug3: send packet: type 1
```
然后使用串口直接连接到板子上，也什么反应都没有。
在此期间，运行的其他的网络服务均可正常使用。

拔电源重启后在 `/var/log/kern.log` 得到的信息如下：
[... 52 lines cut ...]
May 29 06:59:41 orangepizero kernel: [19546.889560] [<c011efa3>] (irq_exit) from [<c015dcd1>] (__handle_domain_irq+0x49/0x84)
May 29 06:59:41 orangepizero kernel: [19546.889563] [<c015dcd1>] (__handle_domain_irq) from [<c05c85e5>] (gic_handle_irq+0x39/0x68)
May 29 06:59:41 orangepizero kernel: [19546.889566] [<c05c85e5>] (gic_handle_irq) from [<c0101ae5>] (__irq_svc+0x65/0x94)
May 29 06:59:41 orangepizero kernel: [19546.889569] Exception stack(0xc0f01f20 to 0xc0f01f68)
May 29 06:59:41 orangepizero kernel: [19546.889572] 1f20: 00000000 01da487c dff5dfb0 c0115841 ffffe000 c0f04f64 c0f04fa8 00000001
May 29 06:59:41 orangepizero kernel: [19546.889576] 1f40: 00000000 c0ec08b0 00000000 c0ff6863 000729a4 c0f01f70 c01075f7 c01075f8
May 29 06:59:41 orangepizero kernel: [19546.889578] 1f60: 40010033 ffffffff
May 29 06:59:41 orangepizero kernel: [19546.889581] [<c0101ae5>] (__irq_svc) from [<c01075f8>] (arch_cpu_idle+0x28/0x2c)
May 29 06:59:41 orangepizero kernel: [19546.889584] [<c01075f8>] (arch_cpu_idle) from [<c013cb7d>] (do_idle+0x12d/0x1b4)
May 29 06:59:41 orangepizero kernel: [19546.889588] [<c013cb7d>] (do_idle) from [<c013cdd9>] (cpu_startup_entry+0x19/0x20)
May 29 06:59:41 orangepizero kernel: [19546.889591] [<c013cdd9>] (cpu_startup_entry) from [<c0e00c17>] (start_kernel+0x3cf/0x3e4)
May 29 06:59:41 orangepizero kernel: [19546.889862] rcu_sched       I    0    10      2 0x00000000
May 29 06:59:41 orangepizero kernel: [19546.889888] [<c0971dd5>] (__schedule) from [<c097216d>] (schedule+0x35/0x84)
May 29 06:59:41 orangepizero kernel: [19546.889899] [<c097216d>] (schedule) from [<c0974595>] (schedule_timeout+0x69/0x264)
May 29 06:59:41 orangepizero kernel: [19546.889917] [<c0974595>] (schedule_timeout) from [<c0169e9f>] (rcu_gp_kthread+0x583/0x79c)
May 29 06:59:41 orangepizero kernel: [19546.889934] [<c0169e9f>] (rcu_gp_kthread) from [<c0131819>] (kthread+0xf9/0x100)
May 29 06:59:41 orangepizero kernel: [19546.889946] [<c0131819>] (kthread) from [<c01010f9>] (ret_from_fork+0x11/0x38)
May 29 06:59:41 orangepizero kernel: [19546.889951] Exception stack(0xd712ffb0 to 0xd712fff8)
May 29 06:59:41 orangepizero kernel: [19546.889959] ffa0:                                     00000000 00000000 00000000 00000000
May 29 06:59:41 orangepizero kernel: [19546.889969] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
May 29 06:59:41 orangepizero kernel: [19546.889976] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
May 29 07:24:47 orangepizero kernel: [21053.048265] printk: systemd-journal: 141 output lines suppressed due to ratelimiting
```

请问此问题有什么引起以及该如何修复。

orangepi-xunlong, 2022-02-08:
你好，光从log是猜不出原因的，你用的哪个镜像，安装了什么软件，做了什么操作都是需要说明白的，另外很重要的一点是此现象能不能稳定复现，如果没法复现是找不出原因以及解决办法来验证的

orangepi-xunlong, 2022-02-09:
你好，Linux5.3.5还是很老的镜像了，目前H3的开发板都已经更新到Linux5.4的系统了，可以试下最新的系统有没有这个问题哈。

closed 2022-03-07 by orangepi-xunlong