orangepi-xunlong/linux-orangepi #9: 6e02ada5 should add audio on h616 - but linux-orangepi kernel says: snd_soc_register_card failed

warpme, 2021-12-22:
Hello,

I just build branch orange-pi-5.13-sunxi64 @ 018ff30bec
By 6e02ada5 I assume it should offer audio support on OrangePi-Zero2 (h616)
Unfortunately audio not works. 
Kernel dmesg reports:

```
dmesg | grep snd
[    1.148241] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    1.428099] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    1.627339] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    1.628555] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    1.629091]   #0: sndahub
[    2.003164] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.065503] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.508424] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.554771] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.556098] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.632447] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
```
 Is this bug or i missed something?

orangepi-xunlong, 2021-12-23:
Hi, 

Is it compiled using orangepi-build？

Or, you can download the images available for HDMI Audio here

http://www.orangepi.org/downloadresources/

warpme, 2021-12-23:
I compiled 5.13 kernel from sources from https://github.com/orangepi-xunlong/linux-orangepi/commit/018ff30bec32c6c0e2e3aed60532699b11bd0daa and use kernel config extracted from Orangepizero2_2.1.6_ubuntu_bionic_desktop_linux5.13.0
Do i need to use any extra code (patches, alsa config etc) to get hdmi audio?

orangepi-xunlong, 2021-12-23:
https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/config/kernel/linux-5.13-sunxi64.config

No patches required, Please try this kernel configuration. Which dts file are you using?

Again: you can use orangepi-build to compile the image for OrangePi Zero2  first and see if there are any problems.

warpme, 2021-12-23:
Thx for quick replay!
This is what i done:
1. build kernel from https://github.com/orangepi-xunlong/linux-orangepi/commit/018ff30bec32c6c0e2e3aed60532699b11bd0daa
2. used kernel config: https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/config/kernel/linux-5.13-sunxi64.config
3. system boots ok but `speaker-test` fails 

Pls find output from:
```
aplay -l
aplay -L
amixer -c0
amixes -c1
amixer -c2
speaker-test -c2
dmesg
```

https://pastebin.com/J44r1A7F

orangepi-xunlong, 2021-12-23:
1. This is the asound.state for Orange Pi Zero2 linux5.13, please make sure all the audio channel is configured correctly.
[ sun50iw9-current](https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/packages/blobs/asound.state/asound.state.sun50iw9-current)

2. Initialize hdmi audio to the available state, audio.wav can be found in our linux5.13 images.
```
aplay /usr/share/sounds/alsa/audio.wav -D hw:0,0 > /dev/null 2>&1
aplay /usr/share/sounds/alsa/audio.wav -D hw:1,0 > /dev/null 2>&1
```

3. The command for playing audio to hdmi is：
```
aplay -D hw:0,0  /usr/share/sounds/alsa/audio.wav
````

warpme, 2021-12-23:
Ok. I got it working :-)
It looks like key thing is to play:
`aplay /usr/share/sounds/alsa/audio.wav -D hw:1,0`
Also i see it must be `audio.wav` file as other files not giving me working sound. 
Is `audio.wav` file contain kind of code required to enable hdmi?

closed 2022-02-08 by orangepi-xunlong