orangepi-xunlong/linux-orangepi #15: Hdmi orangepi 3 lts - next branch

afaulkner420, 2022-04-02:
Is hdmi screen working for orangepi 3 lts under the new 5.16 kernel? 
Im able to build, but when i boot it, the leds do the right thing, but i get a totally black screen. nothing happens.
It seems to work perfectly for Armbian. Which i have now been able to successfully bring up properly. eveything works except bluetooth, crashes when i try to turn it on. Any ideas on that also?

orangepi-xunlong, 2022-04-11:
Hi, Orange Pi 3 LTS Linux5.16 HDMI is fine.

[image]

Can you switch HDMI resolution here？

[image]

afaulkner420, 2022-04-11:
It's totally black/blank from boot. Nothing shows up. I don't get a desktop environment or anything?

afaulkner420, 2022-04-11:
I'm unable to test anything until later today. One issue I have noticed though, is with panfrost. 
121.584328] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x715ec00, tail=0x715ec00
[  121.635491] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x715de00, tail=0x715de00
[  121.637797] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x71724c0, tail=0x71724c0
[  181.666563] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x7171280, tail=0x7171280
[  181.687731] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x7173bc0, tail=0x7173bc0
[  241.728792] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x7172900, tail=0x7172900
[  241.744175] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x711f8c0, tail=0x711f8c0
[  241.750813] panfrost 1800000.gpu: js fault, js=0, status=INSTR_INVALID_PC, head=0x7172f00, tail=0x7172f00

afaulkner420, 2022-04-11:
Not to sure if this is linked at all?

orangepi-xunlong, 2022-04-18:
Sorry, I can't solve GPU-related problems.