orangepi-xunlong/linux-orangepi #17: 5.16-sunxi64: regression breaking audio on other SoCs

warpme, 2022-04-20:
Hi,

I'm playing with https://github.com/orangepi-xunlong/linux-orangepi/commits/orange-pi-5.16-sunxi64 kernel to get audio on h616. 
I also want to use this kernel on other SoCs as it offers audio on h616 SoC.

orange-pi-5.16-sunxi64 works for me on h616, but audio seem to be broken on other SoCs (rockchip and amlogic tested).

Looking on code I see places where sun5i specific code breaks other SOC support:

https://github.com/orangepi-xunlong/linux-orangepi/blob/72895a2481bac46668b478d92e67349a3dda58df/include/sound/soc-dai.h#L96

https://github.com/orangepi-xunlong/linux-orangepi/blob/72895a2481bac46668b478d92e67349a3dda58df/sound/soc/soc-utils.c#L79

https://github.com/orangepi-xunlong/linux-orangepi/blob/72895a2481bac46668b478d92e67349a3dda58df/sound/soc/codecs/hdmi-codec.c#L437

https://github.com/orangepi-xunlong/linux-orangepi/blob/72895a2481bac46668b478d92e67349a3dda58df/sound/soc/codecs/hdmi-codec.c#L478

https://github.com/orangepi-xunlong/linux-orangepi/blob/72895a2481bac46668b478d92e67349a3dda58df/sound/soc/codecs/hdmi-codec.c#L1050

Above code lines are causing audio regression on other SoCs - effectively making 5.16-sunxi64 kernel non-usable with audio apps on non-allwinner SoCs

It is possible to fix this regression?

orangepi-xunlong, 2022-04-21:
You can use a different kernel configuration file to solve this problem.

warpme, 2022-04-21:
_"You can use a different kernel configuration file to solve this problem."_ - indeed. I know this. 
What i want to is: single kernel binary for multiple platforms.
With regression we are discussing here this is impossible....

orangepi-xunlong, 2022-04-22:
The driver needs to be optimized, but you need to do it yourself, I don't have time to do that, thanks!

closed 2022-04-22 by orangepi-xunlong