orangepi-xunlong/linux-orangepi #19: uwe6522 wifi scan fails on 5.16-sunxi64 kernel

warpme, 2022-05-04:
I built xulong 5.16 kernel from sources (72895a248) and have issue with wifi scan on OrangePI Zero2 (uwe5622 wifi chip).
I'm using iwd 1.26.

It looks to me issue is probably in wifi firmware as dmesg says:

```
[  125.323374] WCN: mdbg_assert_read:WCN Assert in connect/source/wifi/rom/mac/cmd/cmd_tx_rom.c line 5111, 0,data length 128
[  125.323445] WCN: stop_loopcheck
[  125.323460] WCN_ERR: chip reset & notify every subsystem...
```

here is log from scan attempt: 

```
root@Myth-Frontend-020049ec9a1a:~ # iwctl
NetworkConfigurationEnabled: disabled
StateDirectory: /var/lib/iwd
Version: 1.26
[iwd]# station wlan0 scan
Operation failed
[iwd]# quit
```

and here is kernel dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.16.17 (piotro@builder-armv8) (aarch64-minimyth-linux-gnu-gcc (GCC) 11.3.0, GNU ld (GNU Binutils) 2.36) #3 SMP PREEMPT Wed May 4 11:51:06 CEST 2022
[    0.000000] Machine model: OrangePi Zero2
[    0.000000] efi: UEFI not found.
[    0.000000] earlycon: uart0 at MMIO32 0x0000000005000000 (options '115200n8')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004007ffff]
[    0.000000]   node   0: [mem 0x0000000040080000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] cma: Reserved 256 MiB at 0x000000006ec00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s78744 r8192 d31848 u118784
[    0.000000] pcpu-alloc: s78744 r8192 d31848 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: root=PARTUUID=346d1a6a-02 rw rootwait earlycon console=ttyS0,115200n8 logo.nologo vt.cur_default=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 735228K/1048576K available (13696K kernel code, 3360K rwdata, 6080K rodata, 6144K init, 547K bss, 51204K reserved, 262144K cma-reserved)
[    0.000000] random: get_random_u64 called from cache_random_seq_create+0x84/0x18c with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 47757 entries in 187 pages
[    0.000000] ftrace: allocated 187 pages with 6 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[... 369 lines cut ...]
[  111.953663] WCN: marlin_bind_verify verify data: 0x016d6ab91e1d071b3ae4604f224021aa
[  111.954254] WCN: check_cp_ready sync val:0xf0f0f0ff, prj_type val:0x0
[  111.954260] sdiohal:sdiohal_runtime_get entry
[  111.954462] WCN: get_cp2_version entry!
[  111.956153] WCN: WCND at cmd read:WCN_VER:Platform Version:MARLIN3_19B_W20.37.4~Project Version:sc2355_marlin3_lite_ott~09-11-2020 21:27:32~
[  111.956170] WCN: switch_cp2_log - close entry!
[  111.957515] WCN: WCND at cmd read:OK
[  111.957563] WCN: then marlin download finished and run ok
[  111.957569] WCN: start_loopcheck
[  111.974634] WCN: get_board_ant_num [one_ant]
[  111.974655] wifi ini path = /lib/firmware/wifi_2355b001_1ant.ini
[  112.034441] unisoc_wifi unisoc_wifi wlan0: mixed HW and IP checksum settings.
[  125.323374] WCN: mdbg_assert_read:WCN Assert in connect/source/wifi/rom/mac/cmd/cmd_tx_rom.c line 5111, 0,data length 128
[  125.323445] WCN: stop_loopcheck
[  125.323460] WCN_ERR: chip reset & notify every subsystem...
[  125.323562] WCN: [marlin_cp2_reset], DO BSP RESET
[  125.323574] WCN: marlin power state:4, subsys: [MARLIN_WIFI] power 0
[  125.323583] WCN: wcn chip start power off!
[  125.323586] sdiohal:sdiohal_runtime_put entry
[  125.323698] sdiohal:sdiohal_runtime_put wait xmit_cnt end
[  125.323710] WCN: chip_power_off
[  125.323714] WCN: marlin chip en dummy pull down -- need manually set GPIO
[  125.323718] WCN: marlin power off!
[  125.323722] WCN: marlin power state:0, subsys: [MARLIN_BLUETOOTH] power 0
```

orangepi-xunlong, 2022-05-19:
Hello, I have updated the firmware and driver for uwe5622, please test it again, thanks!

dmanlfc, 2022-08-11:
wifi works but I have found only for 2.4GHz wifi connections using the latest firmware.
kernel uses - https://github.com/orangepi-xunlong/linux-orangepi/commit/3de14c0e6f2c6c4a8ba67456d3af65bf30c352a8

Example of debug output when trying to connect for 5GHz wifi SSID

sprdwl:Wait disconnect event timeout. [mstime = 38108]
unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection failed status code:45!

andrejusc, 2022-12-02:
@orangepi-xunlong - I'm trying currently such kernel version via latest support Armbian image from Nov 30th:
```
Linux orangepi3-lts 5.15.80-sunxi64 #22.11.1
```
and facing exact same original issue:
```
Dec  2 22:55:15 orangepi3-lts kernel: [  642.341453] WCN: mdbg_assert_read:WCN Assert in connect/source/wifi/rom/mac/cmd/cmd_tx_rom.c line 5111, 0,data length 128
```
and my wifi connection so far doesn't work. Any possible workaround available or any reason there is no yet supported 5.16+ Armbian kernel image downloads here: https://www.armbian.com/orangepi3-lts/ ?

Or any suggestion which of those 6.0.10 images I could try from here: https://github.com/armbian/community /? Tried kinetic/sid, but board doesn't boot then.

closed 2022-12-16 by orangepi-xunlong

dmanlfc, 2022-12-19:
where is the fix @orangepi-xunlong ?

andrejusc, 2022-12-20:
At least for my case of Orange Pi 3 LTS - I've ended up so far with using image `Orangepi3-lts_3.0.8_ubuntu_jammy_desktop_xfce_linux5.16.17.img` from vendor website and WiFi is working with it. Reply to my another question from vendor was that no yet images available for kernel 5.17+. Just as a matter of hands-on - tried to build myself Armbian kernel/image for 6.0.12 with such options as Yes:
```
Unisoc wireless Support (SPARD_WLAN_SUPPORT) [N/y/?] (NEW) y
...
UWE5622 Wi-Fi Driver (WLAN_UWE5622) [N/m/y/?] (NEW) y
```
but that initially resulted in compilation issues (some changes done in `cfg80211_roam_info` definition recently) and then some clash around `do_csum` function. And while fixing such compilation to the extend I understand - later Linker was unhappy with things like such:
```
/root/armbian/cache/toolchain/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: Unexpected GOT/PLT entries detected!
/root/armbian/cache/toolchain/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: Unexpected run-time procedure linkages detected!
/root/armbian/cache/toolchain/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: drivers/net/wireless/uwe5622/unisocwifi/main.o: in function `sprdwl_unregister_wdev':
main.c:(.text+0xe98): undefined reference to `cfg80211_unregister_wdev'
...
```

which is currently yet beyond my knowledge of some other fix/possible workaround.