orangepi-xunlong/linux-orangepi #20: Orange Pi Zero 2 sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed

cf4769, 2022-06-02:
I was told this was the kernel being used for the Batocera 34 image running on the board. Here are two logs after trying to run two games:

log1:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.16.17 (batocera@f52701216d72) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot 2019.11-15034-gc634032f0d) 11.2.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Thu May 26 15:55:55 CEST 2022
[    0.000000] Machine model: OrangePi Zero2
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x7fdcb140-0x7fdccfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004007ffff]
[    0.000000]   node   0: [mem 0x0000000040080000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] cma: Reserved 384 MiB at 0x0000000066c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 19 pages/cpu s37592 r8192 d32040 u77824
[    0.000000] pcpu-alloc: s37592 r8192 d32040 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: initrd=/boot/initrd.gz label=BATOCERA rootwait quiet loglevel=0 console=ttyS0,115200 console=tty3
[    0.000000] Unknown kernel command line parameters "label=BATOCERA", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 612940K/1048576K available (12416K kernel code, 1496K rwdata, 3976K rodata, 2368K init, 383K bss, 42420K reserved, 393216K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: 	Offload RCU callbacks from CPUs: (none).
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000508] Console: colour dummy device 80x25
[    0.000540] printk: console [tty3] enabled
[    0.000608] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000622] pid_max: default: 32768 minimum: 301
[    0.000716] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000729] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002306] rcu: Hierarchical SRCU implementation.
[    0.003471] smp: Bringing up secondary CPUs ...
[    0.004026] Detected VIPT I-cache on CPU1
[    0.004101] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.004660] Detected VIPT I-cache on CPU2
[    0.004721] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.005231] Detected VIPT I-cache on CPU3
[    0.005289] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.005381] smp: Brought up 1 node, 4 CPUs
[    0.005393] SMP: Total of 4 processors activated.
[    0.005399] CPU features: detected: 32-bit EL0 Support
[... 645 lines cut ...]
[    7.452529] random: 7 urandom warning(s) missed due to ratelimiting
[    8.454069] phy_rtl8211f_led_fixup in
[    8.544532] dwmac-sun8i 5020000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[    8.546458] dwmac-sun8i 5020000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[    8.547295] dwmac-sun8i 5020000.ethernet eth0: No Safety Features support found
[    8.547317] dwmac-sun8i 5020000.ethernet eth0: No MAC Management Counters available
[    8.547326] dwmac-sun8i 5020000.ethernet eth0: PTP not supported by HW
[    8.549854] dwmac-sun8i 5020000.ethernet eth0: configuring for phy/rgmii link mode
[    8.591705] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.613774] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.457547] EXT4-fs (mmcblk1p2): recovery complete
[    9.458648] EXT4-fs (mmcblk1p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[    9.585128] squashfs: Unknown parameter 'iocharset'
[   11.366660] Bluetooth: Core ver 2.22
[   11.366798] NET: Registered PF_BLUETOOTH protocol family
[   11.366802] Bluetooth: HCI device and connection manager initialized
[   11.366821] Bluetooth: HCI socket layer initialized
[   11.366829] Bluetooth: L2CAP socket layer initialized
[   11.366845] Bluetooth: SCO socket layer initialized
[   11.430032] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.430058] Bluetooth: BNEP filters: protocol multicast
[   11.430077] Bluetooth: BNEP socket layer initialized
[   12.669740] dwmac-sun8i 5020000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.669797] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   31.708517] cldo1: disabling

dmanlfc, 2022-08-11:
Debain & Ubuntu images from - http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-Zero-2.html as don't produce audio via HDMI.

Both use the 5.16-sunxi64 kernel...

dmanlfc, 2022-08-11:
https://github.com/batocera-linux/batocera.linux/issues/6829

nickyg666, 2022-10-15:
I also have this issue with latest orangepi provided Ubuntu and debian