orangepi-xunlong/linux-orangepi #22: OPi 3 LTS - WIFI crashes after some time

Sineos, 2022-07-28:
SBC: OPi 3 LTS
Linux: Debian Buster
Kernel: Linux orangepi3-lts 5.10.75-sunxi64 #2.1.6 SMP Fri Dec 24 10:19:38 CST 2021 aarch64 GNU/Linux

Stack Trace:
```
[   12.524284] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[172817.049037] ------------[ cut here ]------------
[172817.049095] WARNING: CPU: 0 PID: 23248 at net/wireless/sme.c:757 __cfg80211_connect_result+0x374/0x418 [cfg80211]
[172817.049098] Modules linked in: hci_uart btqca btrtl btbcm btintel bluetooth ecdh_generic ecc zram snd_soc_hdmi_codec sunxi_cir rc_core dw_hdmi_i2s_audio cdc_acm dw_hdmi_cec sun4i_i2s sunxi_cedrus(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc panfrost gpu_sched snd_soc_simple_card snd_soc_simple_card_utils display_connector cpufreq_dt sprdwl_ng cfg80211 sprdbt_tty rfkill uwe5622_bsp_sdio dwmac_sun8i i2c_mv64xxx mdio_mux
[172817.049169] CPU: 0 PID: 23248 Comm: kworker/u8:0 Tainted: G         C        5.10.75-sunxi64 #2.1.6
[172817.049171] Hardware name: OrangePi 3 LTS (DT)
[172817.049186] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[172817.049192] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[172817.049203] pc : __cfg80211_connect_result+0x374/0x418 [cfg80211]
[172817.049213] lr : __cfg80211_connect_result+0x230/0x418 [cfg80211]
[172817.049215] sp : ffff800018cebca0
[172817.049217] x29: ffff800018cebca0 x28: 0000000000000000
[172817.049222] x27: dead000000000100 x26: dead000000000122
[172817.049227] x25: 0000000000000000 x24: ffff000005f64a20
[172817.049232] x23: ffff800018cebce8 x22: ffff00000196a880
[172817.049237] x21: ffff000005f64000 x20: ffff000005f64948
[172817.049241] x19: ffff000008676218 x18: 0000000000000000
[172817.049246] x17: 0000000000000000 x16: 0000000000000000
[172817.049251] x15: 0000007bf489ff6c x14: 0000000000000000
[172817.049256] x13: 0000000000000001 x12: 0000000000000028
[172817.049261] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f
[172817.049266] x9 : 0000000000000001 x8 : 0000000000000012
[172817.049271] x7 : 0000000000000010 x6 : 00000000000000e5
[172817.049275] x5 : 0000000000000004 x4 : 0000000000000000
[172817.049280] x3 : fffffffefd6e0234 x2 : 0000000000000000
[172817.049285] x1 : ffff00000196a880 x0 : 0000000000000000
[172817.049291] Call trace:
[172817.049303]  __cfg80211_connect_result+0x374/0x418 [cfg80211]
[172817.049314]  cfg80211_process_wdev_events+0x1cc/0x250 [cfg80211]
[172817.049325]  cfg80211_process_rdev_events+0x38/0x90 [cfg80211]
[172817.049335]  cfg80211_event_work+0x20/0x38 [cfg80211]
[172817.049343]  process_one_work+0x1c8/0x380
[172817.049347]  worker_thread+0x140/0x4e0
[172817.049353]  kthread+0xf8/0x128
[172817.049358]  ret_from_fork+0x10/0x30
[172817.049362] ---[ end trace 05d485f9ed598a1d ]---
```

dubrsl, 2023-01-15:
I see the same problem on  OPi 4 LTS :(

zhangn1985, 2024-01-09:
yes. this happens on opi-800, with not good wifi signal.