orangepi-xunlong/linux-orangepi #23: cpufreq-dt hang/crash

alxchk, 2022-09-04:
This happens after first cold boot every time. After that any access to cpufreq info causes hang.

```
[    8.777181] Unable to handle kernel paging request at virtual address ffff80000925ec00                                                                                                                                             
[251/579]
[    8.785256] Mem abort info:                                                                                         
[    8.788113]   ESR = 0x96000007                                                                                      
[    8.791212]   EC = 0x25: DABT (current EL), IL = 32 bits                                                            
[    8.795585] systemd[1]: Starting Network Service...                                                                 
[    8.796558]   SET = 0, FnV = 0                                                                                      
[    8.799654]   EA = 0, S1PTW = 0                                                                                     
[    8.802827]   FSC = 0x07: level 3 translation fault                                                                 
[    8.807736] Data abort info:                                                                                        
[    8.810649]   ISV = 0, ISS = 0x00000007                                                                             
[    8.810662]   CM = 0, WnR = 0                                                                                       
[    8.810666] swapper pgtable: 4k pages, 48-bit VAs, pgdp=00000000413bd000                                            
[    8.824225] [ffff80000925ec00] pgd=100000007ffff003, p4d=100000007ffff003, pud=100000007fffe003, pmd=100000007fffa003, pte=0000000000000000                                                                                                 
[    8.838209] Internal error: Oops: 96000007 [#1] SMP                                                                 
[    8.844423] Modules linked in: cpufreq_dt(+) sprdbt_tty rtc_ds1307 sprdwl_ng sunxi_addr cfg80211 fuse nfsd auth_rpcgss nfs_acl lockd grace sunrpc realtek dwmac_sun8i i2c_mv64xxx mdio_mux                                                  
[    8.861077] CPU: 0 PID: 332 Comm: systemd-udevd Not tainted 5.16.17-sun50iw9 #3.0.6                                 
[    8.868740] Hardware name: OrangePi Zero2 (DT)                                                                      
[    8.873187] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)                                         
[    8.880153] pc : cpufreq_generic_get+0x24/0xa8                                                                                                                                                                                              
[    8.884618] lr : cpufreq_online+0x410/0x9d0                                                                         
[    8.888811] sp : ffff80000a5137a0                                                                                   
[    8.892135] x29: ffff80000a5137a0 x28: ffff000002aa3008 x27: 0000000000000001                                       
[    8.899284] x26: ffff000002aa31d0 x25: ffff8000094d8790 x24: ffff80000957c000                                       
[    8.906432] x23: ffff80000943bd48 x22: ffff000002aa3090 x21: 0000000000000000                                       
[    8.913578] x20: 0000000000000000 x19: ffff000002aa3000 x18: 0000000000000000                                       
[    8.920726] x17: 2e726f74616c7567 x16: 65723a726f74616c x15: ffff0000035af240                                       
[    8.927873] x14: ffffffffffffffff x13: ffff000003cccb88 x12: ffff000003cccb86                                       
[    8.935023] x11: 0000000000000030 x10: 0101010101010101 x9 : 0000000000000000                                                                                                                                                               
[    8.942171] x8 : ffff000003cccd48 x7 : ffff000003cccd58 x6 : ffff000002aa30e8                                       
[    8.949318] x5 : 000000000000000f x4 : 000000000000000f x3 : 000000000000000f                                                                                                                                                               
[    8.956464] x2 : ffff800000e22070 x1 : ffff80000925e7c0 x0 : 0000000000000440                                       
[    8.963614] Call trace:                                                                                             
[    8.966069]  cpufreq_generic_get+0x24/0xa8                                                                                                                                                                                                  
[    8.970190]  cpufreq_online+0x410/0x9d0                                                                             
[    8.974037]  cpufreq_add_dev+0xbc/0xd0                                                                              
[    8.977798]  subsys_interface_register+0x9c/0xf8                                                                    
[    8.982432]  cpufreq_register_driver+0x1b0/0x388                                                                    
[    8.987061]  dt_cpufreq_probe+0x200/0x448 [cpufreq_dt]
[    8.992222]  platform_probe+0x68/0xd8
[    8.995897]  really_probe+0xbc/0x428
[    8.999488]  __driver_probe_device+0x114/0x188
[    9.003947]  driver_probe_device+0x80/0x110
[    9.008143]  __driver_attach+0xbc/0x1a0
[    9.011994]  bus_for_each_dev+0x70/0xc0
[    9.015842]  driver_attach+0x24/0x30
[    9.019431]  bus_add_driver+0x140/0x220
[    9.023279]  driver_register+0x64/0x120
[    9.027128]  __platform_driver_register+0x28/0x38
[    9.031843]  dt_cpufreq_platdrv_init+0x24/0x1000 [cpufreq_dt]
[    9.037606]  do_one_initcall+0x54/0x1b8
[    9.041458]  do_init_module+0x44/0x1f0
[    9.045221]  load_module+0x20e0/0x2850
[    9.048982]  __do_sys_finit_module+0xbc/0x128
[    9.053349]  __arm64_sys_finit_module+0x20/0x30
[    9.057889]  invoke_syscall+0x44/0x108
[    9.061650]  el0_svc_common.constprop.0+0xcc/0xf0
[    9.066367]  do_el0_svc+0x24/0x88                                                                                   
[    9.069694]  el0_svc+0x20/0x60                                                                                      
[    9.072755]  el0t_64_sync_handler+0xb0/0xb8                                                                         
[    9.076949]  el0t_64_sync+0x1a0/0x1a4                                                                               
[    9.080628] Code: f0003f81 911f0021 910003fd f8635800 (f8616800)                                                                                                                                                                            
[    9.086726] ---[ end trace 8d7eb1328531e5e4 ]---                                                           
```

ChronosXYZ, 2024-04-23:
Having the same problem. This crash seems very critical, because rebooting is basically broken. systemd cannot shut down processes which are stuck.

ChronosXYZ, 2024-04-23:
> HAd it with orangepizero3 - check your `/boot/armbianEnv.txt (or orangepiEnv.txt)` for `disp_mode=1920x1080p60` <-- seems not to like anything else also please make sure that `/boot/boot.cmd `and `/boot/boot.scr` are aligned, (do not change boot.cmd without recompiling boot.scr) <-- you can always recompile with `sudo mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr`
> 
> I can't say if that was all, but it helped: also please make sure voltage is stable, and avoid having double services active for networking (like networkmanager and systemd-resolved AND ifupdown scripts + avahi-daemon)
> 
> Orange Pi os is based on an old armbian framework, kernel version from vendor is old aswell hope it help, fingers crossed

Thank you, will try that!