orangepi-xunlong/linux-orangepi #33: rk3588 Realtek 8821CU (AC600) causes a boot hang when inserted

Joshua-Riek, 2023-04-01:
Hello, the kernel will hang when inserting a Realtek 8821CU (AC600) WiFi and Bluetooth adapter. This has been observed a few times in the following issues [ubuntu-orange-pi5/#91](https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/91) and  [ubuntu-orange-pi5/#38](https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/38). Some people do not have any issues with this, and some do.

The USB seems to reset continuously, as shown in the below kernel log snippet:
```
[   11.937040] vendor storage:20190527 ret = -1
[   41.450573] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[   72.170556] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[  102.887226] usb 2-1: reset high-speed USB device number 2 using ehci-platform
```

Damenly, 2023-04-05:
It's about usb_modeswitch. When I tried to bring the RTL8821CU on OpenFyde the dmesg occurs too.
Try the instructions at https://github.com/song940/rtl8821cu.

Joshua-Riek, 2023-04-05:
Thank you, it's working now!

closed 2023-04-05 by Joshua-Riek