orangepi-xunlong/linux-orangepi #34: Xunlong Orange Pi R1 Plus Lts   poweroff kernel pannic

William-Tong, 2023-04-07:
[image]
Orange Pi R1 Plus Lts ，在使用poweroff时，内核pannic，无法正常关闭，文件系统无法正常保存，造成每次重启都需要重建ext的文件系统。

William-Tong, 2023-04-07:
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
         Starting Power-Off...
[   68.785401] reboot: Power down
[   69.785831] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[   69.786526] CPU: 0 PID: 1 Comm: systemd-shutdow Tainted: G            E     5.10.44-rockchip64 #2.2.2
[   69.787343] Hardware name: Xunlong Orange Pi R1 Plus Lts (DT)
[   69.787856] Call trace:
[   69.788104]  dump_backtrace+0x0/0x1f0
[   69.788447]  show_stack+0x18/0x28
[   69.788763]  dump_stack+0xc8/0x120
[   69.789084]  panic+0x16c/0x36c
[   69.789378]  do_exit+0xa8c/0xa90
[   69.789685]  __do_sys_reboot+0x170/0x250
[   69.790049]  __arm64_sys_reboot+0x24/0x30
[   69.790427]  el0_svc_common.constprop.0+0x70/0x188
[   69.790869]  do_el0_svc+0x24/0x90
[   69.791182]  el0_svc+0x14/0x20
[   69.791474]  el0_sync_handler+0xb0/0xb8
[   69.791829]  el0_sync+0x160/0x180
[   69.792151] Kernel Offset: disabled
[   69.792475] CPU features: 0x0240002,20002004
[   69.792861] Memory Limit: none
[   69.793161] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000 ]---

baiywt, 2025-04-09:
请问用的哪个镜像？