orangepi-xunlong/linux-orangepi #35: kernel crashes

bbaranoff, 2023-05-07:
It should be a common issue. I will try harder
Whatever I got a good new was able to compile kernel for OPi5 with bifrost and panfrost enabled it should more a pull request but I don't really know how to do :( you can check my modifications here
https://github.com/bbaranoff/bifrost_opi5_patch/
thank you

bbaranoff, 2023-05-10:
Hello again...
I don't got it, it crashes (now) first time was only not booting
you have all the folder here
https://github.com/bbaranoff/OrangePi5-OpenCL/
 Any idea how to get out of this ?

[  662.948714] Modules linked in: zstd sch_fq_codel fuse ip_tables ipv6 dm_mod bifrost_kbase
[  662.949441] CPU: 4 PID: 3094 Comm: Xorg Not tainted 5.10.110+ #1
[  662.949962] Hardware name: Orange Pi 5 (DT)
[  662.950333] pstate: 20400009 (nzCv daif +PAN -UAO -TCO BTYPE=--)
[  662.950859] pc : drm_setversion+0x80/0x18c
[  662.951219] lr : drm_setversion+0x3c/0x18c
[  662.951577] sp : ffffffc01400bc40
[  662.951871] x29: ffffffc01400bc40 x28: ffffffc01400bd58
[  662.952339] x27: 0000000000000010 x26: 0000007fdf21e1e8
[  662.952808] x25: 0000000000000010 x24: ffffff81023bfa00
[  662.953275] x23: ffffff8101c020d0 x22: ffffffc01400bd58
[  662.953744] x21: 0000000000000000 x20: ffffffc01400bd58
[  662.954211] x19: ffffff8101c02000 x18: 0000000000000000
[  662.954680] x17: 0000000000000000 x16: 0000000000000000
[  662.955147] x15: 0000007fdf21e1e8 x14: 0000000000000000
[  662.955615] x13: 0000000000000000 x12: 0000000000000000
[  662.956083] x11: 0000000000000000 x10: 0000000000000000
[  662.956551] x9 : ffffffc010f433f0 x8 : 0000000000000000
[  662.957020] x7 : 0000000000000000 x6 : ffffffc01400bd68
[  662.957488] x5 : ffffffc01400bd68 x4 : 0000000000000000
[  662.957955] x3 : 0000000000000000 x2 : ffffff8103a78f40
[  662.958423] x1 : 0000000000010001 x0 : 0000000000000001
[  662.958891] Call trace:
[  662.959111]  drm_setversion+0x80/0x18c
[  662.959439]  drm_ioctl_kernel+0xb0/0x104
[  662.959787]  drm_ioctl+0x2fc/0x348
[  662.960095]  vfs_ioctl+0x34/0x54
[  662.960379]  __arm64_sys_ioctl+0x84/0xb8
[  662.960729]  el0_svc_common.constprop.0+0x13c/0x1ec
[  662.961154]  do_el0_svc+0x90/0x9c
[  662.961449]  el0_svc+0x20/0x30
[  662.961723]  el0_sync_handler+0xbc/0x158
[  662.962072]  el0_sync+0x1a0/0x1c0

baiywt, 2025-04-09:
@bbaranoff Could you provide the detailed reproduction steps?