orangepi-xunlong/linux-orangepi #37: camera /dev/video* <- not found

youngjae41, 2023-06-22:
model : Orangepi 5B
OS: ubuntu 20.04 focal

camera: OV13850 and 13855 

Hello

When the camera is operated using test_camera.sh, it works normally 

I tried entering all videos 1 to 20 using ffplay /dev/video, but it says no device connected. Can you tell me why?

Siroeshka, 2023-08-15:
Hello, i have the same problem. 
How can i get a frame from OV13850 using c/c++ ? Is there any example?
OpenCV lib cannot open device because it is not connected.

youngjae41, 2023-08-18:
@Siroeshka I found a way. Reboot the orangepi after connecting the cable and enabling the port in orangepi-config and it should work.

Siroeshka, 2023-08-18:
@youngjae41, thank you for reply. unfortunatly this did not helped me.
I found code in test_camera.sh and saw that video11 is used with gst-launch. and it works fine. But OpenCV and ffplay cant view stream. ffplay  says there is no device and failed to load rockchip driver. OpenCV frezzez and do nothing.

I am going to write some code in c++ using gst-launch. Maybe this will help me

youngjae41, 2023-08-21:
@Siroeshka I'm glad it works. When I tested this camera, it only output about 18 fps. So I gave up on using it lol

borisgribkov, 2023-10-30:
Used test_camera.sh, OpenCV with  GStreamer, in all cases frame rate is about 7-8fps even for low resolutions like 320x200. Does anyone have any ideas how to improve? Thanks!

tranhien1612, 2025-03-13:
Hi, i used orangepi5plus with ov13855 camera. i see mainpath video (/dev/video11) and selfpath video (/dev/video12).
```
orangepi@orangepi5plus:~$ v4l2-ctl --list-formats-ext -d /dev/video11
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[1]: 'NV16' (Y/CbCr 4:2:2)
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[2]: 'NV61' (Y/CrCb 4:2:2)
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[3]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[4]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[5]: 'NM21' (Y/CrCb 4:2:0 (N-C))
		Size: Stepwise 32x32 - 4224x3136 with step 8/8
	[6]: 'NM12' (Y/CbCr 4:2:0 (N-C))
		Size: Stepwise 32x32 - 4224x3136 with step 8/8

orangepi@orangepi5plus:~$ v4l2-ctl --list-formats-ext -d /dev/video12
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[1]: 'NV16' (Y/CbCr 4:2:2)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[2]: 'NV61' (Y/CrCb 4:2:2)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[3]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[4]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[5]: 'NM21' (Y/CrCb 4:2:0 (N-C))
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[6]: 'NM12' (Y/CbCr 4:2:0 (N-C))
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[7]: 'GREY' (8-bit Greyscale)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
	[8]: 'RGBP' (16-bit RGB 5-6-5)
		Size: Stepwise 32x32 - 1920x3136 with step 8/8
```

This is some information of media0 and media1:
```
media-ctl -d /dev/media0 -p
Media controller API version 5.10.160

Media device information
------------------------
driver          rkcif
model           rkcif-mipi-lvds2
serial          
bus info        
hw revision     0x0
driver version  5.10.160

Device topology
- entity 1: stream_cif_mipi_id0 (1 pad, 11 links)
            type Node subtype V4L flags 0
            device node name /dev/video0
	pad0: Sink
		<- "rockchip-mipi-csi2":1 [ENABLED]
		<- "rockchip-mipi-csi2":2 []
		<- "rockchip-mipi-csi2":3 []
		<- "rockchip-mipi-csi2":4 []
		<- "rockchip-mipi-csi2":5 []
		<- "rockchip-mipi-csi2":6 []
[... 401 lines cut ...]
             device node name /dev/video17
	pad0: Source
		-> "rkisp-isp-subdev":0 []

- entity 48: rkisp-statistics (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video18
	pad0: Sink
		<- "rkisp-isp-subdev":3 [ENABLED]

- entity 54: rkisp-input-params (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video19
	pad0: Source
		-> "rkisp-isp-subdev":1 [ENABLED]

- entity 60: rkcif-mipi-lvds2 (1 pad, 1 link)
             type V4L2 subdev subtype Unknown flags 0
             device node name /dev/v4l-subdev4
	pad0: Source
		[fmt:SBGGR10_1X10/4224x3136@10000/300000 field:none]
		-> "rkisp-isp-subdev":0 [ENABLED]

```
I see that the size of /dev/video12 is 1920x3136. but i want to change the size of /dev/video12. How to change it?