orangepi-xunlong/linux-orangepi #44: Zero3 wifi正常链接使用中，会有一定概率出错并无法再次链接

evaxige, 2023-09-04:
## 概况
SBC: Zero3 
Kernel：5.4.125
Wifi配置：
```
wpa_cli

> status
bssid=7c:10:c9:xx:xx:xx
freq=5180
ssid=xxx
id=0
mode=station
wifi_generation=5
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
ip_address=192.168.50.241
p2p_device_address=3a:f7:ee:xx:xx:xx
address=38:f7:ee:xx:xx:xx
uuid=3ed8267b-a194-50e3-b76e-2144484xxxxx
ieee80211ac=1
```

## 过程
使用nmcli链接到ap之后，正常使用一段时间，会随机性的出现wifi链接丢失，同时使用nmcli无法再次链接。

dmesg信息如下：
```
[279798.596001] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[280011.375382] ------------[ cut here ]------------
[280011.375411] WARNING: CPU: 0 PID: 5989 at net/wireless/sme.c:756 __cfg80211_connect_result+0x204/0x24c
[280011.375413] Modules linked in: crct10dif_ce ghash_ce sha1_ce sunxi_usbc leds_gpio sprdbt_tty sprdwl_ng uwe5622_bsp_sdio sunxi_rfkill pstore_blk pstore_zone ohci_sunxi ehci_sunxi sunxi_hci
[280011.375437] CPU: 0 PID: 5989 Comm: kworker/u8:2 Not tainted 5.4.125 #1.0.0
[280011.375439] Hardware name: Orange Pi Zero 3 (DT)
[280011.375451] Workqueue: cfg80211 cfg80211_event_work
[280011.375455] pstate: 60400005 (nZCv daif +PAN -UAO)
[280011.375459] pc : __cfg80211_connect_result+0x204/0x24c
[280011.375463] lr : __cfg80211_connect_result+0x1a4/0x24c
[280011.375465] sp : ffffffc015c13ce0
[280011.375466] x29: ffffffc015c13ce0 x28: 0000000000000060 
[280011.375470] x27: 0000000000000000 x26: dead000000000100 
[280011.375473] x25: dead000000000122 x24: 0000000000000000 
[280011.375476] x23: ffffff8038eb0af0 x22: ffffff8038eb0a50 
[280011.375479] x21: ffffff8038eb0a08 x20: ffffff8038e72c18 
[280011.375482] x19: ffffff8038eb0a08 x18: 0000000000000000 
[280011.375485] x17: 0000000000000000 x16: 0000000000000002 
[280011.375487] x15: 00139f3396d60094 x14: 00000442211848a4 
[280011.375490] x13: 000000000000ba7e x12: 0000000000000442 
[280011.375493] x11: 0000000000000000 x10: 0000000000139f33 
[280011.375496] x9 : ffffffc01113b948 x8 : 0000000000000001 
[280011.375499] x7 : 0000000000000010 x6 : 0000000000000002 
[280011.375502] x5 : 0000000000000000 x4 : 000000000000000a 
[280011.375505] x3 : fffffffefbd51b73 x2 : ffffff803cda0000 
[280011.375507] x1 : ffffff803cda0000 x0 : 0000000000000000 
[280011.375511] Call trace:
[280011.375516]  __cfg80211_connect_result+0x204/0x24c
[280011.375519]  cfg80211_process_wdev_events+0x198/0x24c
[280011.375522]  cfg80211_process_rdev_events+0x38/0x90
[280011.375526]  cfg80211_event_work+0x1c/0x30
[280011.375533]  process_one_work+0x1c0/0x384
[280011.375537]  worker_thread+0x16c/0x4f0
[280011.375542]  kthread+0x118/0x120
[280011.375548]  ret_from_fork+0x10/0x18
[280011.375550] ---[ end trace 287536509ab77b6a ]---
```

jscxm, 2024-08-28:
I have the same problem on zero2. Have you solved it

forose, 2024-09-28:
I have the same problem on zero2w. Have you solved it