orangepi-xunlong/linux-orangepi #47: I'm cross-compiling the kernel and I get this error, is this serious? Is it possible to ignore.

lalakii, 2023-09-16:
**log:**
```
aarch64-linux-gnu-ld: Unexpected GOT/PLT entries detected!
aarch64-linux-gnu-ld: Unexpected run-time procedure linkages detected!
aarch64-linux-gnu-ld: ID map text too big or misaligned
aarch64-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ephy_config_init':
sunxi-ephy.c:(.text+0x5f4): undefined reference to `sun50i_ephy_calibrate_value'
aarch64-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ac200_ephy_enable.constprop.0':
sunxi-ephy.c:(.text+0x934): undefined reference to `ac200_enable'
aarch64-linux-gnu-ld: sunxi-ephy.c:(.text+0x95c): undefined reference to `ac200_enable'
aarch64-linux-gnu-ld: sunxi-ephy.c:(.text+0xa14): undefined reference to `sun50i_ephy_calibrate_value'
make[1]: *** [scripts/Makefile.vmlinux:34: vmlinux] Error 1
make: *** [Makefile:1255: vmlinux] Error 2
make: *** Waiting for unfinished jobs....
```

**branch:** orange-pi-6.1-sun50iw9
**device:** orange pi zero3

lalakii, 2023-09-16:
end

closed 2023-09-16 by lalakii

halmartin, 2023-11-16:
@lalakii, could you comment how you resolved the issue? I am facing the same.

qichunren, 2024-01-22:
I go same issue. Did  the change merge into repo?

jscxm, 2025-03-31:
> end

how to esolve