orangepi-xunlong/linux-orangepi #53: orangepizero3: orange-pi-6.1-sun50iw9 branch does not boot, dts missing vcc- supply?

halmartin, 2023-11-18:
Hello,

I am trying to compile 6.1.31 to enable `kvm` since this is not enabled in the default kernel from [leeboby images](https://github.com/leeboby/armbian-images). The kernel is compiled from the branch `orange-pi-6.1-sun50iw9` with `/proc/config.gz` from the armbian image as the base configuration, modified to enable KVM.

The system does not boot with the compiled kernel, because the SD card is not detected in initramfs:
```
Starting kernel ...

Loading, please wait...
Starting systemd-udevd version 252.6-1
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... 1;-1fdone.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Waiting for root file system ... Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=29255239-4f31-4d74-8af8-d56e72fbcab3 does not exist.  Dropping to a shell!
(initramfs) dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.31-custom (root@042ab3c5c237) (aarch64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Bin
utils for Ubuntu) 2.34) #2 SMP Fri Nov 17 10:51:24 UTC 2023
[    0.000000] Machine model: OrangePi Zero3
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000040000000-0x000000013fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x13f7bf040-0x13f7c0fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004007ffff]
[    0.000000]   node   0: [mem 0x0000000040080000-0x000000013fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000013fffffff]
[    0.000000] cma: Reserved 128 MiB at 0x00000000f8000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.4
[    0.000000] percpu: Embedded 20 pages/cpu s41704 r8192 d32024 u81920
[... 335 lines cut ...]

There are no such errors during boot in the armbian kernel/image provided in [leeboby images](https://github.com/leeboby/armbian-images). The `sun50i-h616-orangepi-zero2.dts` includes `&pio` with a [mapping of the supply-vcc to PMIC](https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-6.1-sun50iw9/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero2.dts#L269-L275). No such mapping exists in `sun50i-h616-orangepi-zero3.dts`.

I tried to add the appropriate mapping based on the Zero3 schematic, but board does not output anything over UART after handover from u-boot:
```
--- a/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero3.dts
+++ b/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero3.dts
@@ -218,6 +219,14 @@ reg_dldo1: ldo2 {
        };
 };

+&pio {
+        vcc-pc-supply = <&reg_dldo1>;
+        vcc-pf-supply = <&reg_dldo1>;
+        vcc-pg-supply = <&reg_aldo1>;
+        vcc-ph-supply = <&reg_aldo1>;
+        vcc-pi-supply = <&reg_aldo1>;
+};
+
 &uart0 {
        pinctrl-names = "default";
        pinctrl-0 = <&uart0_ph_pins>;
```

It seems like the repository is not matching the binaries shipped, otherwise I would expect a kernel with the same `.config` to boot. Could someone please take a look?

broadblues, 2023-11-18:
Which model of orangepi zero 3 do you have? If the 4Gb you will need to replace both the uboot and the dtb as it needs a different ones from those generated for  2 & 4 Gb builds by the orangepi build system. There a choice of two board types I'm guessing it should be 3!

There are instructions from the link you gave above IIRC. Apply those steps to your custom image. 

I don't know what the exact issue is. I have decompiled that 4gb dtb and patched it to enable Mali GPU.

broadblues, 2023-11-18:
https://github.com/leeboby/opizero3-uboot-dtb in chinese so use google translate if you can't read it.

halmartin, 2023-11-18:
> Which model of orangepi zero 3 do you have? If the 4Gb you will need to replace both the uboot and the dtb as it needs a different ones from those generated for 2 & 4 Gb builds by the orangepi build system.

Yes, I am aware of this and have already done it. The stock images boot fine. Compiling your own kernel with the same configuration, for me, does not boot.

broadblues, 2023-11-18:
Hmm I started from the configs provided in the orangepi-build/next branch and then applied the fixes described to get a booting image. It would be better if it would "just work" for all variants of the board.

halmartin, 2023-11-18:
> I started from the configs provided in the orangepi-build/next branch and then applied the fixes described to get a booting image

Could you link to the repository and fixes you're describing?

The [orangepi-build](https://github.com/orangepi-xunlong/OrangePi_Build) base repository has not been updated in years.

broadblues, 2023-11-18:
https://github.com/orangepi-xunlong/orangepi-build/tree/next