orangepi-xunlong/linux-orangepi #57: uwe5622-unisocwifi: wait_for_completion() causes bugos system load and dmesg spam

magicnat, 2023-12-20:
[`wait_for_completion`](https://github.com/orangepi-xunlong/linux-orangepi/blob/c192384ac705014cedbdf7fb5b1949922e6d2f1f/drivers/net/wireless/uwe5622/unisocwifi/tx_msg.c#L318) puts the calling thread in the "uninterruptible sleep" state, making the kernel count the CPU utilization as 100%, leading to the bugos 1.0 system load.

Other reports of the same issue:

- http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=10454
- https://forum.armbian.com/topic/31654-orange-pi-zero-2w/
- https://github.com/MichaIng/DietPi/issues/6594#issuecomment-1831724676

a quick dirty fix is to make the above wait interruptible by changing the `tx_down` implementation:

```c
/* seam for tx_thread */
void tx_down(struct sprdwl_tx_msg *tx_msg)
{
	int ret;
	while (1) {
		ret = wait_for_completion_interruptible(&tx_msg->tx_completed);
		if (ret == -ERESTARTSYS) {
			continue;
		}

		return;
	}
}
```

this makes the thread "sleeps" properly when it waits:

```
% ps auxw | egrep '[S]PRDWL_TX_QUEUE'
root         375  0.0  0.0      0     0 ?        S<   08:27   0:00 [SPRDWL_TX_QUEUE]
% uptime
 09:16:34 up 49 min,  2 users,  load average: 0.00, 0.00, 0.00
```

this also fix the spam of `protocol 0000 is buggy, dev wlan0` message for me (similar report [here](https://github.com/orangepi-xunlong/linux-orangepi/issues/50) and [here](https://www.linux.org/threads/protocol-0000-is-buggy-dev-wlan0.42774/)). my suspension is that `wait_for_completion` somehow returned without the semaphore actually being completed, causing some bogus behavior down the line.

rohoog, 2024-02-18:
I had the same issue on 6.1.31 with opi zero 2w and was also experiencing wifi stability issues, sometimes freezing the whole system. I applied your patch and now the system load is normal, the system 'feels' more responsive when accessing through ssh from wifi and I haven't seen a lockup anymore.

rohoog, 2024-02-22:
Although it seems a lot more stable, I still get errors on the wifi after a while. Dmesg contains:
```
[169232.347916] unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection sm_state (5), status: (2)!
[169232.347971] unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection SX551D9FAAA failed status code:1!
[169238.493047] sprdwl:sprdwl_fc_add_share_credit, 541, mode:1 closed, index:0, share it
```

jscxm, 2024-09-14:
> Although it seems a lot more stable, I still get errors on the wifi after a while. Dmesg contains:
>
> ```
> [169232.347916] unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection sm_state (5), status: [...]

Can you reconnect the wifi after disconnecting

jscxm, 2024-09-14:
> Although it seems a lot more stable, I still get errors on the wifi after a while. Dmesg contains:
>
> ```
> [169232.347916] unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection sm_state (5), status: [...]

Can you reconnect the wifi after disconnecting？

rohoog, 2024-09-15:
No, the whole wifi device is inoperational in networkmanager. The only way to get it working again (apart from hard rebooting) is to hook a serial port, log in, unload the kernel modules and reload them and kicking networkmanager to re-enumerate the wifi adapters.
I am considering to create a cron job that checks regularly if the wifi is still operational and if it isn't, do the unloading/reloading automatically. This is very ugly and will still cause a network interruption every now and then.

forose, 2024-09-28:
> No, the whole wifi device is inoperational in networkmanager. The only way to get it working again (apart from hard rebooting) is to hook a serial port, log in, unload the kernel modules and reload [...]

I have the save problem can you tell me how to reload

forose, 2024-09-28:
> No, the whole wifi device is inoperational in networkmanager. The only way to get it working again (apart from hard rebooting) is to hook a serial port, log in, unload the kernel modules and reload [...]

yes! thank you