orangepi-xunlong/linux-orangepi #58: orangepi zero3 linux kernel 6.1.31 Compilation error

lazybat, 2023-12-23:
使用官方build报错：

```
drivers/net/ethernet/allwinner/sunxi-gmac.c:2193:13: error: ‘ucc_geth_suspend’ undeclared here (not in a function)
drivers/net/ethernet/allwinner/sunxi-gmac.c:2194:13: error: ‘ucc_geth_resume’ undeclared here (not in a function)
make[5]: *** [scripts/Makefile.build:250: drivers/net/ethernet/allwinner/sunxi-gmac.o] Error 1
make[4]: *** [scripts/Makefile.build:500: drivers/net/ethernet/allwinner] Error 2
make[3]: *** [scripts/Makefile.build:500: drivers/net/ethernet] Error 2
make[2]: *** [scripts/Makefile.build:500: drivers/net] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2
make: *** [Makefile:2012: .] Error 2
[ error ] ERROR in function compile_kernel [ build.sh:305 -> scripts/main.sh:630 -> scripts/main.sh:533 -> scripts/compilation.sh:515 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

```

taspenwall, 2024-01-01:
I'm having a hard time building this kernel as well I have the following 
`aarch64-linux-gnu-ld: Unexpected GOT/PLT entries detected!
aarch64-linux-gnu-ld: Unexpected run-time procedure linkages detected!
aarch64-linux-gnu-ld: ID map text too big or misaligned
aarch64-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ephy_config_init':
sunxi-ephy.c:(.text+0x5fc): undefined reference to `sun50i_ephy_calibrate_value'
aarch64-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ac200_ephy_enable.constprop.0':
sunxi-ephy.c:(.text+0x944): undefined reference to `ac200_enable'
aarch64-linux-gnu-ld: sunxi-ephy.c:(.text+0x96c): undefined reference to `ac200_enable'
aarch64-linux-gnu-ld: sunxi-ephy.c:(.text+0xa24): undefined reference to `sun50i_ephy_calibrate_value'
make[4]: *** [scripts/Makefile.vmlinux:34: vmlinux] Error 1
make[3]: *** [Makefile:1255: vmlinux] Error 2
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [debian/rules:6: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:86: bindeb-pkg] Error 2
make: *** [Makefile:1656: bindeb-pkg] Error 2                                                     /1`

Not sure what I'm doing wrong.  I cross compiling from my x86 dev machine.

qichunren, 2024-01-22:
I got same issue with @taspenwall

legut2, 2024-07-07:
`orange-pi-6.1-sun50iw9` doesn't compile correctly. Something wrong with `drivers/net/phy/sunxi-ephy.c` here is my verbose output:
```
2024-07-06T15:07:16.2466302Z + [ arm64 = um ]
2024-07-06T15:07:16.2466675Z + wl=
2024-07-06T15:07:16.2468387Z + ld=/home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld
2024-07-06T15:07:16.2471589Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn
2024-07-06T15:07:16.2473743Z + ldlibs=
2024-07-06T15:07:16.2475912Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds
2024-07-06T15:07:16.2478008Z + [ .tmp_vmlinux.kallsyms1 != 1 ]
2024-07-06T15:07:16.2480417Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds --strip-debug
2024-07-06T15:07:16.2482565Z + is_enabled CONFIG_VMLINUX_MAP
2024-07-06T15:07:16.2483323Z + grep -q ^CONFIG_VMLINUX_MAP=y include/config/auto.conf
2024-07-06T15:07:16.2488950Z + /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld -EL -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds --strip-debug -o .tmp_vmlinux.kallsyms1 --whole-archive vmlinux.a .vmlinux.export.o init/version-timestamp.o --no-whole-archive --start-group ./drivers/firmware/efi/libstub/lib.a --end-group
2024-07-06T15:07:16.9260057Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: Unexpected GOT/PLT entries detected!
2024-07-06T15:07:16.9265631Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: Unexpected run-time procedure linkages detected!
2024-07-06T15:07:16.9269440Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: ID map text too big or misaligned
2024-07-06T15:07:17.1645527Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ephy_config_init':
2024-07-06T15:07:17.1649926Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:163:(.text+0x5f4): undefined reference to `sun50i_ephy_calibrate_value'
2024-07-06T15:07:17.1654148Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ac200_ephy_enable':
2024-07-06T15:07:17.1658743Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:354:(.text+0x900): undefined reference to `ac200_enable'
2024-07-06T15:07:17.1664467Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:357:(.text+0x928): undefined reference to `ac200_enable'
2024-07-06T15:07:17.1671988Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:382:(.text+0x9f0): undefined reference to `sun50i_ephy_calibrate_value'
2024-07-06T15:07:17.5438801Z make[2]: *** [scripts/Makefile.vmlinux:34: vmlinux] Error 1
2024-07-06T15:07:17.5441061Z make[1]: *** [Makefile:1255: vmlinux] Error 2
2024-07-06T15:07:17.5441789Z make[1]: *** Waiting for unfinished jobs....
2024-07-06T15:07:17.9968652Z make[1]: Leaving directory '/home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom'
2024-07-06T15:07:17.9972188Z make: *** [package/pkg-generic.mk:283: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/.stamp_built] Error 2
```

I diffed kernel.org's 6.1.31 and 6.1.31 provide in `orange-pi-6.1-sun50iw9` just to see if I could spot anything that stood out:
```
[... cut ...]
Only in linux-6.1.31/include/linux/mfd: ac200.h
Only in linux-6.1.31/scripts/dtc/include-prefixes/arm64/allwinner: overlay
Only in linux-6.1.31/scripts/dtc/include-prefixes/arm64/allwinner: sun50i-h616-cpu-opp.dtsi
Only in linux-6.1.31/scripts/dtc/include-prefixes/arm64/allwinner: sun50i-h616-orangepi-zero2w.dts
Only in linux-6.1.31/scripts/dtc/include-prefixes/arm64/allwinner: sun50i-h616-orangepi-zero3.dts
Only in linux-6.1.31/sound/soc/sunxi: sun50iw9-codec.c
Only in linux-6.1.31/sound/soc: sunxi_v2
```

baiywt, 2025-04-09:
It builds fine for me. 
[image]