orangepi-xunlong/linux-orangepi #59: orangepi zero2 - kernel 6.1 (next) - broken wlan and i2c overlay (?)

alxchk, 2023-12-23:
Tried with a8686ef79c020c4070068b020e86e0d87a309409

Built with defaults using build.sh

Installed:
```
linux-dtb-next-sun50iw9_3.1.0_arm64.deb
linux-headers-next-sun50iw9_3.1.0_arm64.deb
linux-image-next-sun50iw9_3.1.0_arm64.deb
linux-libc-dev_3.1.0_arm64.deb
```

At boot wlan failed with trace:
```
[    7.508333] WCN: start_marlin [MARLIN_WIFI]
[    7.508425] Unable to handle kernel NULL pointer dereference at virtual address 00000000000001d0
[    7.508442] Mem abort info:
[    7.508450]   ESR = 0x0000000096000004
[    7.508460]   EC = 0x25: DABT (current EL), IL = 32 bits
[    7.508474]   SET = 0, FnV = 0
[    7.508484]   EA = 0, S1PTW = 0
[    7.508494]   FSC = 0x04: level 0 translation fault
[    7.508506] Data abort info:
[    7.508513]   ISV = 0, ISS = 0x00000004
[    7.508523]   CM = 0, WnR = 0
[    7.508533] user pgtable: 4k pages, 48-bit VAs, pgdp=0000000044d8a000
[    7.508548] [00000000000001d0] pgd=0000000000000000, p4d=0000000000000000
[    7.508577] Internal error: Oops: 0000000096000004 [#1] SMP
[    7.515419] Modules linked in: sprdwl_ng(+) sunxi_addr nfsd cfg80211 auth_rpcgss rfkill nfs_acl lockd grace rtc_ds1307 libcomposite fuse dm_mod sunrpc btrfs realtek dwmac_sun8i mdio_mux
[    7.532050] CPU: 0 PID: 332 Comm: systemd-modules Tainted: G                T  6.1.31-sun50iw9 #3.1.0
[    7.541283] Hardware name: OrangePi Zero2 (DT)
[    7.545733] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    7.552706] pc : get_loopcheck_status+0x10/0x18
[    7.557259] lr : start_marlin+0x80/0xe4
[    7.561112] sp : ffff80000a07b7c0
[    7.564433] x29: ffff80000a07b7c0 x28: ffff800001010300 x27: 0000000000000000
[    7.571591] x26: 0000000000000000 x25: 0000000000000000 x24: ffff800009406bf8
[    7.578746] x23: 000000000000003a x22: 0000000000000000 x21: ffff800008d8da50
[    7.585902] x20: ffff800008d8de30 x19: 0000000000000002 x18: 0000000000000000
[    7.593058] x17: 6674616c703d4d45 x16: 5453595342555300 x15: ffff0000097f8ec0
[    7.600214] x14: 00000000000001cc x13: ffff0000097f8fd0 x12: 00000000ffffffea
[    7.607370] x11: 00000000ffffbfff x10: 00000000ffffbfff x9 : ffff00003fc812c0
[    7.614525] x8 : 000000000005ffe8 x7 : c0000000ffffbfff x6 : 000000000002fff4
[    7.621681] x5 : 000000000015ffa8 x4 : 0000000000000000 x3 : ffff80000a07b568
[    7.628836] x2 : db7b5ba3afe33600 x1 : 0000000000000000 x0 : 0000000000000000
[    7.635992] Call trace:
[    7.638446]  get_loopcheck_status+0x10/0x18
[    7.642643]  sprdwl_probe+0x20/0x348 [sprdwl_ng]
[    7.647412]  platform_probe+0x68/0xd8
[    7.651093]  really_probe+0xc0/0x3a8
[    7.654689]  __driver_probe_device+0x7c/0x160
[    7.659064]  driver_probe_device+0x3c/0x120
[    7.663266]  __device_attach_driver+0xbc/0x158
[    7.667727]  bus_for_each_drv+0x78/0xc8
[    7.671579]  __device_attach+0xf8/0x1b0
[    7.675432]  device_initial_probe+0x14/0x20
[    7.679634]  bus_probe_device+0x9c/0xa8
[    7.683487]  device_add+0x3c8/0x770
[    7.686992]  platform_device_add+0x104/0x270
[    7.691274]  unisoc_wlan_init+0xa0/0x1000 [sprdwl_ng]
[    7.696454]  do_one_initcall+0x54/0x1d8
[    7.700308]  do_init_module+0x48/0x1d0
[    7.704072]  load_module+0x1984/0x1e48
[    7.707834]  __do_sys_finit_module+0xc4/0x140
[    7.712204]  __arm64_sys_finit_module+0x20/0x30
[    7.716748]  invoke_syscall+0x44/0x108
[    7.720516]  el0_svc_common.constprop.0+0x44/0xf0
[    7.725238]  do_el0_svc+0x2c/0xc8
[    7.728571]  el0_svc+0x2c/0x88
[    7.731641]  el0t_64_sync_handler+0xb8/0xc0
[    7.735840]  el0t_64_sync+0x18c/0x190
[    7.739524] Code: 900067e0 d503233f d50323bf f9419c00 (b941d000)
[    7.745629] ---[ end trace 0000000000000000 ]---
```

Also there is no i2c-3 device

[config.gz](https://github.com/orangepi-xunlong/linux-orangepi/files/13759721/config.gz)
[dmesg.txt](https://github.com/orangepi-xunlong/linux-orangepi/files/13759722/dmesg.txt)
[orangepiEnv.txt](https://github.com/orangepi-xunlong/linux-orangepi/files/13759723/orangepiEnv.txt)

baiywt, 2025-04-09:
Do you encounter this issue when using the official prebuild images？

AGM1968, 2025-04-10:
The panic is intriguing, it looks like sdio is de-referenced, possibly due to missing firmware .ini, perhaps extract it from the official image or as suggested test the official image to see if present.