orangepi-xunlong/linux-orangepi #65: kernel image install fail

tester58126, 2024-01-29:
http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_5_Plus#Linux_Development_Manual

I followed the URL above to build the kernel on my Orange Pie 5 Plus board and tried to proceed with the update, but the `sudo make install` command failed.

[image]

tester58126, 2024-01-29:
it can't seem to generate a proper symlink, is there a workaround ?

rohoog, 2024-02-17:
Maybe the target filesystem doesn't support symlinks? (FAT/VFAT)

cyring, 2025-01-04:
With Ubuntu or Debian, I build for debian packages within `linux_orangepi` branch `rk3588`
Those packages, Image, Headers and DTB are created into the parent directory.

```sh
# Clone the repo
cd linux_orangepi
git pull
# Find within that kernel tree and copy the byteshift patch into /tmp
# make help for exact syntax
make -j8 bindeb
# Go to parent and search for 3 latest packages
cd ..
ls -lt
# Install from privilege account 
su
dpkg -i <image.deb> <headers.deb> <dtb.deb>
# Before rebooting check in /boot that image and initrd files have been copied: headers in /lib/module 
```