orangepi-xunlong/linux-orangepi #66: Inquiry on Persistent Memory Page State Bugs on Orange Pi

archalchemy, 2024-02-04:
Dears Orangepi Developers,

I am reaching out for insights on a recurring issue I’ve encountered with my Orange Pi. These errors are not necessarily triggered by high memory loads and occur unexpectedly, I consistently face a "BUG: Bad page state" error. This issue manifests across both Ubuntu and Debian kernels, leading to system instability.

The error messages appear as follows:

BUG: Bad page state in process memtester ...

Given the persistence and nature of these messages, I am uncertain whether this is a hardware-related fault or a bug within the kernel itself. Notably, the system is equipped with 16 GB of RAM, and the errors occur despite successful passes of memory tests through memtester.

I would greatly appreciate any guidance or experiences you could share regarding this issue. Should you suspect a hardware defect, would you recommend any specific diagnostic steps to confirm this? Conversely, if this is indicative of a kernel bug, are there known patches or workarounds that could resolve these errors?

Thank you for your time and assistance.

Best regards,

[image]

[image]

[image]

ValdikSS, 2025-02-27:
If you use Wi-Fi, make sure that the antenna is located far away from the RAM. Locate is as far from the board as possible.

baiywt, 2025-04-09:
@archalchemy Which image are you using for testing? Could you please share the debug log from the serial console?