orangepi-xunlong/linux-orangepi #71: linux6.1.43内核系统编译后无法开机

tom-xingshuo, 2024-03-21:
板子：orangepi5B
系统：官方最新debian或者Ubuntu，linux内核6.1.43
内核源码：https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.1-rk35xx
在板子上编译命令：
tar zxf orange-pi-6.1-rk35xx.tar.gz
mv orange-pi-6.1-rk35xx linux-orangepi
cd linux-orangepi
git pull origin orange-pi-6.1-rk35xx
make rockchip_linux_defconfig
make -j10 
sudo make modules_install
sudo make install 
sudo make dtbs_install INSTALL_DTBS_PATH=/boot/dtb/
sudo reboot
重启后无法点亮屏幕，红灯常亮但是绿灯不亮

sdrzmgy, 2024-04-07:
一样的问题

DreamLand7707, 2024-05-14:
加一，重启之后就寄了

baiywt, 2025-02-17:
串口log发下看看

stars-under, 2025-02-28:
[5bNewKernel.log](https://github.com/user-attachments/files/19023026/5bNewKernel.log)
reboot后无法启动,卡在Starting kernel ...

baiywt, 2025-04-07:
@stars-under @tom-xingshuo 内核配置文件用这个https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-rockchip-rk3588-current.config

覆盖掉orange-pi-6.1-rk35xx/arch/arm64/configs/rockchip_linux_defconfig