orangepi-xunlong/linux-orangepi #73: orangepi 3b 可以用rt-linux内核吗

Coloryr, 2024-04-16:
我想使用rt-linux
看到了这个仓库里面有5.10版的针对rk35xx的rt内核
但是编译的时候出现
```
arch/arm64/boot/dts/rockchip/rk3566-orangepi-cm4.dts:1290.14-1299.11: Warning (graph_port): /rkisp-vir1/port: graph node unit address error, expected "0"
make[2]: *** [scripts/Makefile.build:273: kernel/sched/core.o] Error 1
make[1]: *** [scripts/Makefile.build:516: kernel/sched] Error 2
make: *** [Makefile:1938: kernel] Error 2
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:639 -> scripts/main.sh:542 -> scripts/compilation.sh:515 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```
这个内核可以使用吗

baiywt, 2025-02-17:
目前不支持

TongXu2, 2025-07-01:
什么时候可以支持一下吗

TongXu2, 2025-07-01:
> 目前不支持

啥时候能支持一下呀