orangepi-xunlong/linux-orangepi #75: Orange Pi OS (Arch) for Orange Pi 5 Plus comes with no pacman mirror

benoitrolland, 2024-05-08:
It is impossible to install or update any software on an Orange pi without a repo. 
the default http://nl.mirror.orangeinfra.online/ is not responding.
Changing it inside /etc/pacman.d/mirrorlist to standard available repos leeds to errors because pacman is looking for community.db, core.db and extra.db files that seems to be missing on such mirrors.
How should one use pacman on OPIOS ?

charlie34, 2024-05-14:
I have the same problem

charlie34, 2024-05-14:
I need this fix urgent because I am developing a new solution with this computer and I need to install some packages

jlsilicon, 2025-05-02:
Useless supt.
Same problem here.
Why do peopled put this Trashware out ... ?
Orange Pi pushes this on its site, but we can't update it or use it.
Really sad !

irorange27, 2025-05-11:
I met the same question.
So bad.