orangepi-xunlong/linux-orangepi #76: Idle system having high load average

abasu0713, 2024-05-10:
Hi OrangePi team,

I am compiling a full Linux Image and loading it on a board with some extra Kernel Modules (Ceph and RBD).
It's fresh install and has no software installed on it. Not even an update. Can you guys tell me why am I seeing a load average of 1.0 min all the time? and there are hundreds of processes for zswap. 

```bash
orangepi@orangepi3b:~$ uptime
 00:58:32 up 47 min,  3 users,  load average: 1.02, 1.01, 1.00
```

[image]

[image]

I don't have this issue on an OrangePi 5B or 5Plus with the same kernel modules self-compiled. 

[image]

[image]

It's a simple Ubuntu Server Jammy. 

```bash
orangepi@orangepi3b:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
orangepi@orangepi3b:~$ uptime
 01:02:30 up 51 min,  3 users,  load average: 1.07, 1.02, 1.00
```

I have even tried turning off swap. To no avail. Would really appreciate any help

ValdikSS, 2025-02-27:
That's because some of the custom driver/kernel modifications use kernel threads which are treated as 'not idle' by loadavg.
Usually this happens when `wait_event()` is used, which flags the thread as TASK_UNINTERRUPTIBLE, which is counted as 'busy' by loadavg. The CPU is not under load with this code style, just an incorrect loadavg reporting.

abasu0713, 2025-02-27:
> That's because some of the custom driver/kernel modifications use kernel threads which are treated as 'not idle' by loadavg.
> Usually this happens when `wait_event()` is used, which flags the thread as TASK_UNINTERRUPTIBLE, which is counted as 'busy' by loadavg. The CPU is not under load with this code style, just an incorrect loadavg reporting.
> 
> It could be `sdiohal_tx_thread` which you see in top.

Yeah. I couldn't find the right reasons. I moved to Debian server and used armbian. And have no such issues there. Been super stable as well.

baiywt, 2025-04-07:
We've tested with Orangepi3b_1.0.8_ubuntu_jammy_desktop_xfce_linux5.10.160.img and the load average appears to be normal on our side.

[image]