orangepi-xunlong/linux-orangepi #77: pi5 pro 更换成实时内核后没有网口设备

QinHaifeiCN, 2024-05-11:
1、安装了官方提供Orangepi5pro_1.0.2_debian_bookworm_desktop_xfce_linux5.10.160.img的系统
2、git clone https://github.com/orangepi-xunlong/linux-orangepi.git --branch=orange-pi-5.10-rk35xx-rt
3、复制[https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-rockchip-rk3588-legacy-rt.config] 到.config
4、添加.config配置   CONFIG_NET_VENDOR_MOTORCOMM=y  CONFIG_FUXI=y

大致按如上编译出实时内核，但是切换成这个之后重新启动系统。发现网口设备没有识别

QinHaifeiCN, 2024-05-13:
官方的处理太慢了，还是需要靠自己解决。。 
1、git clone https://github.com/orangepi-xunlong/linux-orangepi.git --branch=orange-pi-5.10-rk3588      
2、将上述库中的/drivers/net/ethernet/motorcomm/ 文件夹复制到实时内核源码的同一个位置
3、更改实时内核源码/drivers/net/ethernet/Makefile ，添加一项obj-y += motorcomm/

上面三个步骤已经解决网口设备无法识别的问题

这个还会有wifi设备无法识别的问题，需要更改make时候的.config文件。需要添加一些项
CONFIG_NET_VENDOR_MOTORCOMM=y
CONFIG_FUXI=y
CONFIG_R8125=m
CONFIG_SSV6200_CORE=m
CONFIG_BCMDHD_SDIO=y

yangquanll, 2024-09-25:
确实技术支持太弱了，Armbian 构建太麻烦了，对于系统定制

baiywt, 2025-04-07:
@chiyu-Charon 请使用最新的代码。
https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-5.10-rk35xx-rt
https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-rockchip-rk3588-legacy-rt-opi5pro.config