orangepi-xunlong/linux-orangepi #81: Joystick doesn't detected in /dev/input

sreesms, 2024-06-25:
I have to use the joystick for control of a robot, I expect to get /dev/input/js0 after connecting the Joystick, but js0 is appearing.
[image]

 When I enter lsusb I get like this:

[image]

below is the screenshot of dmesg after disconnecting and connecting the joystick usb. 
[image]

USB joystick is working fine in another PC.

Anyone help how to solve this issue?

broadblues, 2024-06-26:
I would guess your joystick is added as a evdev device rather than the older joystick specific.

Look in 

/dev/input/by-id

do you see something like your joystick there?

try running 

evtest

that will show the different devices attached to those /dev/input/event* files

sreesms, 2024-06-27:
> I would guess your joystick is added as a evdev device rather than the older joystick specific.
> 
> Look in
> 
> /dev/input/by-id
> 
> do you see something like your joystick there?
> 
> try running
> 
> evtest
> 
> that will show the different devices attached to those /dev/input/event* files

ls /dev/input/by-id doesn't show joystick, evtest also doesn't show joystick events.

RoboticsChen, 2024-07-10:
I encountered the same problem，and i tried modify the kenal options, then re-compile and install kernal manually，but it didn't work @orangepi-xunlong

RoboticsChen, 2024-07-23:
I rebuilt the kernel and enabled joystick support. 
https://github.com/RoboticsChen/orangepi_kernel.git