orangepi-xunlong/linux-orangepi #84: 树莓派3B 内核6.6版本 编译debian和ubuntu都无法启动

tianfeng-yang, 2024-07-19:
commit: ef82e033e45b07c9c2baff487b957d3995c8db33
所有参数都是用默认，编译出来的镜像烧录之后，黑屏，无法启动，SPI也清除了，依旧无法启动。绿灯不闪烁。

orangepi-build/external/config/boards/orangepi3b.conf 中添加BOOT_FDT_FILE="rockchip/rk3566-orangepi-3b.dtb"  编译的debian bookworm能启动，分配不了IP地址。

百度网盘下载的Ubuntu镜像是正常的，但kernel是5.10的。。。。。

tianfeng-yang, 2024-07-19:
新买的开发板， 硬件版本是v2.1的，看起来这个硬件版本压根还没有支持kernel6.6, 所有文档都没有标注。。。。。

umifly, 2024-07-26:
目前v1.1的硬件也无法通过nvme启动新版本固件，看起来这个1.0.6版本的固件本身就有问题，两个月前已经反馈过了，但是官方管生不管养

tianfeng-yang, 2024-07-30:
> 目前v1.1的硬件也无法通过nvme启动新版本固件，看起来这个1.0.6版本的固件本身就有问题，两个月前已经反馈过了，但是官方管生不管养

我现在跑是跑起来了，但遇到另一个问题，烧录到EMMC之后，首次重启必然无法启动，需要用工具切换一下存储到EMMC，感觉像uboot默认从SPI启动？你这个可能也是同样的问题，用工具切换一下试试。

tianfeng-yang, 2024-07-31:
[修改了一下，Kernel6.6基本功能已经能跑起来了](https://github.com/tianfeng-yang/orangepi-build.git)

baiywt, 2025-04-07:
测试最新的代码编译出的镜像启动（nvme）没有问题，请问你spi flash的程序下的官网最新的吗？[image]

umifly, 2025-04-07:
> [@umifly](https://github.com/umifly) 测试最新的代码编译出的镜像启动（nvme）没有问题，请问你spi flash的程序下的官网最新的吗？[image]

之前是遇到过这个问题，现在我手上没有开发板，测试不了