orangepi-xunlong/linux-orangepi #86: Orange Pi 5 Max Device Tree doesn't have &av1d

DummyTitan, 2024-09-02:
Hello,

I'm opening this issue to observe if someone is willing or able to help fix this issue. Whenever I attempt to use transcoding in Jellyfin with the proper Mali drivers, any codec gets transcoded except the AV1. A team member from Jellyfin @nyanmisaka stated that she verified the device tree and didn't have the module for decoding added, like in this example: https://github.com/armbian/linux-rockchip/commit/d06d87ee891e6ea16c9500a386c3a2705d80ce27 so that I should communicate to attempt fixing this. I even tried fixing it myself, but unfortunately, I am unable. Please help.

DummyTitan, 2024-09-03:
Just in case, there is also an issue with the updates, at least in Debian, which has broken kernel essential files causing the system to crash and not be able to start Debian again unless reinstalled.

baiywt, 2025-04-07:
Hi, thanks for reporting this.

The issue has been resolved in the following commit:
https://github.com/orangepi-xunlong/linux-orangepi/commit/5b622f37c74d893f8a16a9d236fc4152abb6129a