orangepi-xunlong/linux-orangepi #93: orangepi 5plus 以太网内核崩溃

13037639065, 2025-01-22:
问题描述：插着网线开机，系统内核有如下日志。
硬件环境：Hardware name: RK3588 OPi 5 Plus (DT)
系统环境：5.10.160-rockchip-rk3588 #1.0.8
```
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896225] ------------[ cut here ]------------
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896244] NETDEV WATCHDOG: enP3p49s0 (r8125): transmit queue 1 timed out
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896312] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:467 dev_watchdog+0x194/0x1bc
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896317] Modules linked in: xt_REDIRECT iptable_filter rtc_sd3078 can_raw rfcomm can overlay iptable_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 l2tp_ppp l2tp_core ip6_udp_tunnel udp_tunnel pppox ppp_generic slhc bnep ch341 btusb btrtl btbcm btintel bluetooth pwm_fan iwlmvm iwlwifi rockchip_canfd can_dev sch_fq_codel option usb_wwan usbserial ip_tables ipv6 pgdrv r8125
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896441] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.160-rockchip-rk3588 #1.0.8
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896444] Hardware name: RK3588 OPi 5 Plus (DT)
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896451] pstate: 60400009 (nZCv daif +PAN -UAO -TCO BTYPE=--)
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896456] pc : dev_watchdog+0x194/0x1bc
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896460] lr : dev_watchdog+0x194/0x1bc
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896464] sp : ffffffc00a16bd70
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896467] x29: ffffffc00a16bd70 x28: 0000000000000100
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896474] x27: ffffffc008dfe138 x26: ffffffc00a16be60
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896481] x25: ffffffc00a03e140 x24: 0000000000000001
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896487] x23: ffffffc00a051140 x22: ffffffc008dfe138
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896493] x21: ffffff8109000498 x20: 0000000000000001
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896500] x19: ffffff8109000000 x18: 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896506] x17: 0000000000000000 x16: 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896513] x15: 000000000000000a x14: 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896520] x13: 0000000000000000 x12: 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896527] x11: 0000000000000000 x10: 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896533] x9 : ffffffc0080de574 x8 : 6575712074696d73
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896540] x7 : 6e617274203a2935 x6 : ffffffc00a0c78a7
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896546] x5 : ffffff82f6d029c8 x4 : 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896552] x3 : 0000000000000027 x2 : 0000000000000000
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896558] x1 : 0000000000000000 x0 : ffffffc009d6b640
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896566] Call trace:
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896572]  dev_watchdog+0x194/0x1bc
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896582]  call_timer_fn+0x94/0x138
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896586]  __run_timers+0x1c4/0x1f0
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896591]  run_timer_softirq+0x2c/0x50
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896598]  __do_softirq+0x220/0x2c4
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896606]  __irq_exit_rcu+0x4c/0x9c
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896610]  irq_exit+0x18/0x28
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896617]  __handle_domain_irq+0xc4/0xc8
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896621]  gic_handle_irq+0x2b8/0x308
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896626]  el1_irq+0xcc/0x180
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896636]  arch_local_irq_enable+0xc/0x28
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896641]  cpuidle_enter+0x40/0x58
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896647]  do_idle+0x23c/0x260
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896651]  cpu_startup_entry+0x2c/0x44
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896659]  rest_init+0xd0/0xe4
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896665]  arch_call_rest_init+0x18/0x24
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896670]  start_kernel+0x584/0x5bc
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896674]
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896674] PC: 0xffffffc008dfe2cc:
Jan 17 16:27:47 2582404102982M6 kernel: [  368.896676] e0cc  b943ca80 6b15001f 54000108 a94153f3 a9425bf5 a94363f7 f94023f9 a8c57bfd
```

13037639065, 2025-01-22:
相关资料： https://github.com/torvalds/linux/commit/95ecba62e2fd201bcdcca636f5d774f1cd4f1458

13037639065, 2025-01-25:
https://gbmc.googlesource.com/linux/+/8d3ea3d402db94b61075617e71b67459a714a502

baiywt, 2025-04-07:
[image]
我测试没出现这个问题，两个网口都测过。