orangepi-xunlong/linux-orangepi #95: orangepi5b hdmi 输出颜色异常

zmoth, 2025-02-18:
在使用orangepi5b接小米电视时显示颜色异常。与 https://github.com/Joshua-Riek/ubuntu-rockchip/issues/121 一样的问题，在启动后执行xrandr --output HDMI-1 --set "color_format" "rgb"能够使得颜色恢复正常，颜色错误的原因是使用了ycbcr444的颜色。但是在系统启动显示logo的阶段颜色还是错误的。

G4ZZAR-Software, 2025-03-12:
You have no idea how much you have just helped me with my own project, as a linux novice this single line of code has just fixed a project I am working on. Thank you so much for sharing

zmoth, 2025-03-13:
> You have no idea how much you have just helped me with my own project, as a linux novice this single line of code has just fixed a project I am working on. Thank you so much for sharing

Hello, if you also want to fix the system startup display logo phase color issue. This is one of my methods:

1. First you must download the [official kernel](https://github.com/orangepi-xunlong/linux-orangepi.git) and compile it 
2. Find  [`kernel/orange-pi-6.1-rk35xx/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c`](https://github.com/orangepi-xunlong/linux-orangepi/blob/fb528a6014381c12a129e4f5e33c8034d46ad25e/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c#L2030)
3. Forces the value in `*color_format` to be `RK_IF_FORMAT_RGB`; (`*color_format = RK_IF_FORMAT_RGB`)