orangepi-xunlong/linux-orangepi #98: Orange Pi Zero 3 Wi-Fi module issue (uwe5622)

ValdikSS, 2025-02-27:
Hello, Orange Pi!

The Zero 3 board which comes with AW859A / 20U5622 Wi-Fi module (uwe5622 driver) has a bug. After some time, any Wi-Fi connection stops working with the following in the journal:

```
wpa_supplicant[25499]: wlan0: Trying to associate with xx:xx:xx:xx:xx:xx (SSID='xxx' freq=2452 MHz)
wpa_supplicant[25499]: FT: Invalid key management type (2)
wpa_supplicant[25499]: wlan0: Association request to the driver failed
```

And the following in dmesg:

```
[259241.515808] unisoc_wifi wlan0: sprdwl_report_connection xxx failed status code:30!
[345665.483530] unisoc_wifi wlan0: sprdwl_report_connection sm_state (5), status: (2)!
[345665.483567] unisoc_wifi wlan0: sprdwl_report_connection xxx failed status code:1!
```

And it does not fix on reconnecting. The only way is to reboot the board or to remove/insert the kernel module again.  
I believe this is a uwe5622 driver bug. Could you please check for driver updates from them? Or could you ask Unisoc to fix the issue?

I'm running kernel 6.1 with the uwe5622 driver from your repository.

broadblues, 2025-03-03:
Interesting , I have seen that the Zero 3 will drop wifi connection, and then not reconnect, usually takes some days to occur here

krinxiao, 2025-03-07:
I am also encountering the same issue. It appears that the official team is aware of this bug, but there are currently no plans to fix it. I have heard that Armbian includes several patches for the 5622, so I wonder if switching to Armbian might resolve the problem.

krinxiao, 2025-03-08:
I’m currently using `modprobe -r sprdwl_ng` followed by `modprobe sprdwl_ng` as a temporary workaround, and I’ve added these commands to a cron job. I’m still testing to see if this improves overall stability. It’s a real shame about this board; its cost-performance ratio is excellent, but the official team doesn’t seem to care much about maintaining the drivers and device tree—they’re just focused on selling the hardware.

ValdikSS, 2025-05-15:
The issue is related to 802.11r roaming.

After some time _something_ happens which results in instant association reject to the network from the driver itself. I've enabled debug output, there doesn't seem to be any queries to the hardware upon association request when this happens.

1. 802.11r is honored by the driver even if wpa_supplicant `key_mgmt` does not include `FT-PSK`, that's the reason for `FT: Invalid key management type (2)` messages.  
Could be fixed with `wpa_cli 
3. There's in-hardware roaming offload feature which probably doesn't work correctly
4. There's a possibility that the driver version does not correspond to the chip firmware version

I've tried to disable hardware offload with `priv->fw_capa &= ~SPRDWL_CAPA_11R_ROAM_OFFLOAD;`, and enable software-only roaming, it still fails. It can't even roam manually, with `wpa_cli roam bssid`, it fails the same way, and after some attempts it fails to connect back to the same network which was working before.

ValdikSS, 2025-05-19:
I've contacted Cdtech, Orange Pi, and Unisoc, and haven't received any reply yet.

If anyone wants to dig deeper, the firmware of the module is a Cortex-M thumb code. Not encrypted or obfuscated.

YipKo, 2026-08-21:
Hi, I investigated this issue and opened a possible fix here:

https://github.com/armbian/uwe5622/pull/17

The problem appears to occur when the firmware reports `SPRDWL_ROAM_SUCCESS` while the driver is still in the `SPRDWL_CONNECTING` state. The original code does not treat this as a successful connection, so the driver's state can remain stuck. Subsequent connection attempts may then keep failing until the module is reloaded or the board is rebooted.

The patch handles this state transition and also corrects the cleanup behavior for failed connection and roaming attempts.

I rebuilt and tested the patched `sprdwl_ng` module on an Orange Pi Zero 3. So far, I have not seen the same stuck connection state again.

If possible, could you rebuild `sprdwl_ng` from the PR branch for your exact kernel version and test it on your device? The module should be built against the matching kernel source/configuration. Please let me know whether the issue still occurs and attach the relevant `dmesg` logs if it does.

ValdikSS, 2026-09-04:
Roaming fixed, thanks @YipKo!
https://github.com/armbian/uwe5622/pull/17

closed 2026-09-04 by ValdikSS