orangepi-xunlong/linux-orangepi #99: OrangePi 5b安装PREEMPT_RT实时补丁后，wifi模块加载错误

chiyu-Charon, 2025-03-24:
使用仓库https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-5.10-rk35xx-rt和https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-rockchip-rk3588-legacy-rt.config安装了实时补丁。
备份替换原来的内核
cp vmlinuz-5.10.160-rt89-rt-rockchip-rk3588 /boot/Image
cp initrd.img-5.10.160-rt89-rt-rockchip-rk3588 /boot/uInitrd
cp arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5b.dtb /boot/dtb/rockchip/
update-initramfs -u -k 5.10.160-rt89-rt-rockchip-rk3588

重启后发现wifi模块无法加载。
ip link show只有eth0有线网卡，没有wlan0。

查看发现dhd出了问题
[    7.441496] [dhd] dhd_conf_set_path_params : Final fw_path=fw_bcm43752a2_pcie_ag.bin
[    7.441502] [dhd] dhd_conf_set_path_params : Final nv_path=nvram_AP6275P.txt
[    7.442839] [dhd] dhd_os_get_img_fwreq: request_firmware fw_bcm43752a2_pcie_ag.bin err: -2
[    7.442846] [dhd] _dhdpcie_download_firmware:4813 dongle image file download failed
[    7.442851] [dhd] dhd_bus_start: failed to download firmware fw_bcm43752a2_pcie_ag.bin
/lib/firmware/ap6275p/目录下是有对应固件。

尝试软连接
固件是找到了
[dhd] dhd_os_get_img_fwreq: fw_bcm43752a2_pcie_ag.bin (936074 bytes) open success
[dhd] dhd_os_get_img_fwreq: nvram_AP6275P.txt (7458 bytes) open success
但是PCIe识别错误
[dhd] dhdpcie_pci_probe: PCIe Enumeration failed
[dhd] dhdpcie_dongle_attach : failed to read PCI configuration space!

allenwang2, 2025-04-02:
我也遇到了同样的问题， @chiyu-Charon 你有解决了吗

baiywt, 2025-04-07:
用最新的代码试试看，https://github.com/orangepi-xunlong/linux-orangepi/commits/orange-pi-5.10-rk35xx-rt
[image]

chiyu-Charon, 2025-11-07:
> [@chiyu-Charon](https://github.com/chiyu-Charon) [@allenwang2](https://github.com/allenwang2) 用最新的代码试试看，https://github.com/orangepi-xunlong/linux-orangepi/commits/orange-pi-5.10-rk35xx-rt [image]

@baiywt 你好，我根据最新的仓库进行编译测试，编译完成后wifi模块仍然加载错误

[image]

我的操作流程如下，我没有在UbuntuPC上交叉编译，直接在香橙派上进行编译：
`git clone https://github.com/orangepi-xunlong/linux-orangepi.git --branch=orange-pi-5.10-rk35xx-rt`
`https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-rockchip-rk3588-legacy-rt-opi5pro.config `
下载配置文件将其放到linux-orangepi文件夹下，修改为.config文件，不修改参数只改了gcc编译工具跑基本测试，直接编译 `make -j$(nproc)`
编译完成后先备份了原先的内核，之后`make INSTALL_MOD_STRIP=1 modules_install`和`make install`。我看终端输出，现在的install.sh脚本会直接更新uInitrd和initramfs。之后重启设备，确认了实时补丁生效，但是wifi模块仍然有问题。我将编译后的新的设备树进行替换也不行，尝试自己下载wifi相关驱动，但是没找到正确方法。
请问是哪个步骤出了问题呢，应该如何修复，谢谢，没有wifi我要ssh远程连非常麻烦。