orangepi-xunlong/linux-orangepi #101: orangepi cm4 DSI custom panel doesnt read by DRM system

BMValeev, 2025-04-03:
If I try to add panel, driver panel-simple-dsi is loaded ok, but DRM writes "failed to found panel or bridge -19" Seems problem in the endpoints, but there is no info how to add it correctly.

baiywt, 2025-04-07:
@BMValeev Hi, The error failed to find panel or bridge -19 typically indicates that the DRM subsystem couldn't find a valid panel or bridge node. This is often caused by incorrect or missing endpoint links between the DSI controller and the panel in the device tree.
To fix this, make sure:
1. The DSI controller node (&dsi0) has a ports block with an endpoint that links to the panel.
2. The panel node has a ports block and an endpoint that links back to the DSI controller.
Here’s a reference example:
&dsi0 {
    status = "okay";
    ports {
        port@1 {
            reg = <1>;
            dsi_out: endpoint {
                remote-endpoint = <&panel_in>;
            };
        };
    };

    panel@0 {
        compatible = "simple-panel-dsi";
        reg = <0>;
        backlight = <&backlight>;
        pinctrl-names = "default";
        pinctrl-0 = <&lcd_enable_gpio>, <&lcd_rst_gpio>;
        reset-gpios = <&gpio4 RK_PA5 GPIO_ACTIVE_LOW>;
        reset-delay-ms = <500>;
        enable-delay-ms = <60>;
        prepare-delay-ms = <60>;
        ...

        ports {
            port@0 {
                reg = <0>;
                panel_in: endpoint {
                    remote-endpoint = <&dsi_out>;
                };
            };
        };
    };
};

Let me know if you need help reviewing your DTS.

BMValeev, 2025-04-13:
I tried your way, but it doesnt adapted to  DRM rockchip, due to driver init sequence as I understod.
For now I modified raspi driver 
Results next:

> dmesg | grep drm;echo "" ;dmesg | grep dsi ; echo "1"; dmesg | grep wave
[    5.491963] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[    5.493088] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[   10.378528] [drm] Initialized rknpu 0.9.2 20231018 for fde40000.npu on minor 0
[   17.216597] dw-mipi-dsi-rockchip fe070000.dsi: [drm:dw_mipi_dsi_host_attach] *ERROR* Failed to find panel or bridge: -517
[   17.220741] rockchip-vop2 fe040000.vop: [drm:vop2_bind] vp0 assign plane mask: 0x3f, primary plane phy id: 4
[   17.220752] rockchip-vop2 fe040000.vop: [drm:vop2_bind] vp1 assign plane mask: 0x0, primary plane phy id: -1
[   17.220758] rockchip-vop2 fe040000.vop: [drm:vop2_bind] vp2 assign plane mask: 0x0, primary plane phy id: -1
[   17.220919] rockchip-vop2 fe040000.vop: [drm:vop2_bind] Cluster0-win0 as cursor plane for vp0
[   17.220988] [drm] failed to init overlay plane Cluster0-win1
[   17.221072] rockchip-drm display-subsystem: bound fe040000.vop (ops 0xffffffc0091d81c0)
[   17.221241] rockchip-drm display-subsystem: bound fe070000.dsi (ops 0xffffffc0091e7da0)
[   17.223386] rockchip-drm display-subsystem: failed to parse loader memory
[   17.228725] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 800x1280p60, type: 16(if:40, flag:0x0) for vp0 dclk: 70000000     
[   17.518886] dw-mipi-dsi-rockchip fe070000.dsi: [drm:dw_mipi_dsi_bridge_enable] final DSI-Link bandwidth: 936 x 2 Mbps
[   17.578050] rockchip-drm display-subsystem: [drm] fb0: rockchipdrmfb frame buffer device
[   17.589659] [drm] Initialized rockchip 3.0.0 20140818 for display-subsystem on minor 1
[   17.123904] panel_waveshare_dsi_v2: loading out-of-tree module taints kernel.
[   17.124730] wave_my_dr 1-0045: dsi panel: waveshare,my_driver
[   17.146933] wave_my_dr 1-0045: dsi:waveshare panel hw id = 0x1
[   17.172491] wave_my_dr 1-0045: dsi:waveshare panel size = 101
[   17.182497] wave_my_dr 1-0045: dsi:waveshare panel mcu version = 0x1
[   17.216174] wave_my_dr 1-0045: dsi:start0
[   17.216303] wave_my_dr 1-0045: dsi:start1
[   17.216463] wave-ts-dsi fe070000.dsi.0: dsi:init
[   17.216597] dw-mipi-dsi-rockchip fe070000.dsi: [drm:dw_mipi_dsi_host_attach] *ERROR* Failed to find panel or bridge: -517
[   17.216603] wave-ts-dsi fe070000.dsi.0: dsi: failed to attach dsi to host: -517
[   17.216607] wave-ts-dsi fe070000.dsi.0: dsi:success
[   17.218197] wave_my_dr 1-0045: dsi:start2
[   17.218208] wave_my_dr 1-0045: dsi:start3
[   17.218213] wave_my_dr 1-0045: dsi:start4
[   17.220346] wave-ts-dsi fe070000.dsi.0: dsi:init
[   17.221241] rockchip-drm display-subsystem: bound fe070000.dsi (ops 0xffffffc0091e7da0)
[   17.223434] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   17.228316] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   17.302207] wave_my_dr 1-0045: waveshare dsi prepare start:
[   17.518834] wave_my_dr 1-0045: waveshare dsi prepare:
[   17.518866] wave_my_dr 1-0045: waveshare dsi enable:
[   17.518886] dw-mipi-dsi-rockchip fe070000.dsi: [drm:dw_mipi_dsi_bridge_enable] final DSI-Link bandwidth: 936 x 2 Mbps
[   17.589687] wave-ts-dsi fe070000.dsi.0: dsi:success
[   17.592988] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   18.135302] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   18.334980] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   18.351911] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   29.889413] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   29.895298] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   44.272879] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   47.238470] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1
[   47.256273] wave_my_dr 1-0045: waveshare dsi:get_mode: DSI-1

DTS file:

`/*
 * vc4-kms-dsi-ili9881-5inch-overlay.dts
 */

/dts-v1/;
/plugin/;

/ {
	fragment@3 {
		target = <0xffffffff>;

		__overlay__ {
			status = "okay";
		};
[... 126 lines cut ...]
		};
	};
	//__overrides__ {
	//	dsi1 = <&dsi_frag>, "target:0=",<&dsi1>, <&i2c_frag>, "target:0=", <&i2c1>;
	//};

	__fixups__ {
		dsi1 = "/fragment@3:target:0";
		i2c1 = "/fragment@2:target:0";
		video_phy1 = "/fragment@4:target:0"; //enable added
		dsi1_in_vp0 = "/fragment@5:target:0"; //disable added
		dsi1_in_vp1 = "/fragment@6:target:0"; //disable added
		route_dsi1 = "/fragment@7:target:0"; //disable added
		hdmi = "/fragment@8:target:0"; //disable added
		dsi1_out_panel = "/fragment@9:target:0"; //disable added
		edp = "/fragment@11:target:0";
		panel_in_edp = "/fragment@12:target:0";
		wave_panel = "/fragment@13:target:0";
		panel_wave_in_dsi1 = "/fragment@14:target:0";
	};
};
`

Driver there:
https://github.com/BMValeev/malloc_check/blob/main/driver_testc.c