orangepi-xunlong/linux-orangepi #111: orangepi5 ultra 5.10/6.1内核hdmirx无法使用

Pure-Zzzz, 2025-07-03:
[image]

khusika, 2025-07-16:
> [image]

There is a fix for this issue in the upstream [rockchip-kernel](https://github.com/rockchip-linux), or maybe you can apply [this commit](https://github.com/rockchip-linux/kernel/commit/4bfb13e85d62b8c307e69cc1249c0f69f5d51134) over [orange-pi-6.1-rk35xx](https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.1-rk35xx) branch

Alternatively you can use my branch below which has merged the upstream kernel from linux-stable and rockhip-kernel
`https://github.com/khusika/linux_stable-orangepi/tree/orange-pi-6.1-rk35xx`
and make sure that you disable interactive cpu governor from your kernel config (`# CONFIG_CPU_FREQ_GOV_INTERACTIVE is not set`) as this is broken with the latest linux-stable API

Pure-Zzzz, 2025-07-17:
@khusika Thank you for your help! I've tested the updated kernel on both Orange Pi 5 Plus and 5 Ultra. While the patch works perfectly on the Plus, it doesn't appear to take effect on the Ultra. This might indicate some remaining compatibility issues with the Ultra's official support. I've noticed similar reports in community discussions ([Armbian](https://forum.armbian.com/topic/49741-orange-pi-5-ultra-support/), [Radxa](https://forum.radxa.com/t/hdmi-input-support/25740)).