orangepi-xunlong/linux-orangepi #115: 6.1.43没有实时rt的内核吗？在kernel找不到6.1.43

mrtian, 2025-07-17:
如题，我尝试使用6.1.46的patch来编译，报错：error: implicit declaration of function ‘serial8250_set_IER’; did you mean ‘serial8250_set_THRI’

wentywenty, 2025-09-18:
我也在找

wentywenty, 2025-09-18:
drivers/tty/serial/8250/8250_core.c:303:17: error: implicit declaration of function ‘serial8250_set_IER’; did you mean ‘serial8250_set_THRI’? [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: drivers/tty/serial/8250/8250_core.o] Error 1
drivers/tty/serial/8250/8250_port.c:755:17: error: implicit declaration of function ‘serial8250_set_IER’; did you mean ‘serial8250_set_THRI’? [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: drivers/tty/serial/8250/8250_port.o] Error 1
make[4]: *** [scripts/Makefile.build:500: drivers/tty/serial/8250] Error 2
make[3]: *** [scripts/Makefile.build:500: drivers/tty/serial] Error 2
make[2]: *** [scripts/Makefile.build:500: drivers/tty] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2
make: *** [Makefile:2012: .] Error 2
[ error ] ERROR in function compile_kernel [ build.sh:305 -> scripts/main.sh:641 -> scripts/main.sh:544 -> scripts/compilation.sh:506 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 

一样的问题！

wentywenty, 2025-09-18:
> 如题，我尝试使用6.1.46的patch来编译，报错：error: implicit declaration of function ‘serial8250_set_IER’; did you mean ‘serial8250_set_THRI’

中间patch的时候有个文件找不见：can't find file to patch at input line 7570
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/net/devlink/leftover.c b/net/devlink/leftover.c
|index f1c6bd727a83..f6b660604e29 100644
|--- a/net/devlink/leftover.c
|+++ b/net/devlink/leftover.c
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored

wentywenty, 2025-09-24:
> 如题，我尝试使用6.1.46的patch来编译，报错：error: implicit declaration of function ‘serial8250_set_IER’; did you mean ‘serial8250_set_THRI’

https://github.com/wentywenty/orangepi_RT_patch

mrtian, 2025-09-25:
感谢感谢